ev3dev/ev3dev #1032: DNS freeze in Brickman UI

rickyzhang82, 2018-01-22:
### What are you trying to do?
Manually change DNS in Brickman UI for tethering

### What did you expect to happen?
DNS setup works

### What actually happened?
Brickman UI freeze once I clicked enter

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

### Workarounds
I manually add DNS by modifying /etc/resolve.conf

dlech, 2018-01-22:
Can you give exact steps to repeat the problem? When you say "tethering" do you mean you are using the tethering feature in brickman?

rickyzhang82, 2018-01-22:
@dlech This is an unrelated question. Why the change in /etc/resolve.conf is not persisted after server restarts? I confirm that rootfs is mounted on `/` is rw.

dlech, 2018-01-22:
We are using ConnMan for network management. It "owns" `/etc/resolv.conf`. So as a workaround, you should set the DNS using `connmanctl` instead of editing `/etc/resolv.conf` directly.

OtacilioNeto, 2019-04-01:
How can workaround this bug?

OtacilioNeto, 2019-04-07:
For me this command works
connmanctl config wifi_e84e063cf28d_4469616e6120636861636f6e_managed_psk --ipv6 off --ipv4 dhcp --nameservers 8.8.8.8 8.8.4.4

Minecreator200, 2023-03-12:
**For everyone having the same problem:**

1. Connect via SSH to your EV3 running ev3dev
2. Run `ls /var/lib/connman/`
3. Copy the name of the target network device from the list
4. Run `connmanctl config <copied network device from before> --nameservers <your nameserverlist>`

Important: Replace `<your nameserverlist>` with a list of namerserver IPs seperated by spaces.

Example: `connmanctl config wifi_12341234_123412341234 --namerservers 1.1.1.1 8.8.8.8 8.8.4.4`