ev3dev/ev3dev #1033: Docker Cross Compiler Images missing python interpreter

tcwan, 2018-01-24:
### What are you trying to do?
Build ev3dev-c libraries inside ev3dev/debian-*-cross Docker container

### What did you expect to happen?

The build process to complete

### What actually happened?

Missing python interepreter inside Docker container

### What hardware and software are you using (including version numbers)?

ev3dev/debian-*-cross Docker images (latest)

### Comment

Having python included in the cross image may be useful for supporting various projects? Maybe it should be a different image from the default.
I will try to come up with a Dockerfile for this.

dlech, 2018-01-24:
IMHO, writing a Dockerfile to install per-project build dependencies is standard operating procedure. The base Docker images are bare-bones to keep the download size at a minimum.

tcwan, 2018-01-24:
Ok.

[debian-stretch-cross-pytools.dockerfile](https://github.com/tcwan/docker-cross/blob/python-tools/ev3dev-stretch/debian-stretch-cross-pytools.dockerfile)

I'm stuck trying to download the Docker image though. 
I've logged into my Docker account.

TCWiMac:docker tcmac$ docker build .
Sending build context to Docker daemon   2.56kB
Step 1/2 : FROM ev3dev:debian-stretch-cross
pull access denied for ev3dev, repository does not exist or may require 'docker login'

dlech, 2018-01-24:
`ev3dev:debian-stretch-cross` colon should be a slash.

tcwan, 2018-01-24:
I changed it to a slash, but now it complains about permissions.
The new docker build seems to assume non-root user in my case. How did the original build succeed?

```
$ docker build .
Sending build context to Docker daemon   2.56kB
Step 1/2 : FROM ev3dev/debian-stretch-cross
 ---> 1ac15970696e
Step 2/2 : RUN apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install --yes --no-install-recommends         python         scons
 ---> Running in f1ef4f6238d4
W: chmod 0700 of directory /var/lib/apt/lists/partial failed - SetupAPTPartialDirectory (1: Operation not permitted)
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
The command '/bin/sh -c apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install --yes --no-install-recommends         python         scons' returned a non-zero code: 100
```

tcwan, 2018-01-24:
I can fix it by prepending `sudo` to `apt`, but is this the correct solution?

dlech, 2018-01-24:
The `ev3dev/debian-stretch-cross` image [sets the user](https://github.com/ev3dev/docker-cross/blob/master/ev3dev-stretch/debian-stretch-cross.dockerfile#L32) to `compiler`. So, either you need to change the user back to `root` or use `sudo`. Either is OK.

The docker image file format is `<org>/<image>:<tag>`. If `:<tag>` is omitted, then it is assumed to be `:latest`. If `<org>:` is omitted, then it is assumed to be an "official" docker image. So in your case, when you wrote `ev3dev:debian-stretch-cross`, docker was looking for the `latest` tag of an "official" image named `ev3dev`. And there is not such an image, so it correctly gave you an error. When you write `ev3dev/debian-stretch-cross`, docker looks for the `latest` tag of an image named `debian-stretch-cross` in the `ev3dev` organization. When you write `debian:stretch`, docker looks for the `stretch` tag of an "official" image named `debian`.

tcwan, 2018-01-24:
Got it. I've created a [pull request](https://github.com/ev3dev/docker-cross/pull/3) for the new dockerfile.

closed 2018-01-25 by tcwan