ev3dev/ev3dev #1034: does ev3dev support a wifi dongle with chipset ralink 7601?

mariomolfese, 2018-01-29:
I'm trying to make work a lego ev3 mindstorm robot booted with an ev3dev sdcard (so it runs ev3dev linux, indeed. Brickman 0.8.1), making it use a wifi dongle with chipset ralink 7601. The robot boots, linux is ok, but wifi does not work (it says "wi-fi not available")

Everything went well with a wifi dongle with chipset ralink RT3070 (that I've lost, I don't have it anymore)

With a wifi dongle with chipset ralink 7601 i can't see anything in the wi-fi sections: the display simply says to me "wi-fi not available"

How can I make ev3dev work with the wifi dongle with chipset ralink 7601?

thanks in advance to everyone

dlech, 2018-01-29:
Can you connect to the EV3 with USB or Bluetooth and run `lsusb` in an SSH command line and paste the results here?

mariomolfese, 2018-01-29:
yes:
believing it was a problem of drivers, in the meanwhile I upgraded the sdcard to snapshot-ev3dev-stretch-ev3-generic-2017-12-13. Then I booted with that version (brickman 0.10.0) but the result has not changed. I logged in via bluetooth, following your advice, and now I paste you the following result

login as: robot
Using keyboard-interactive authentication.
Password:
Linux ev3dev 4.9.58-ev3dev-1.6.0-ev3 #1 PREEMPT Tue Oct 31 22:16:58 CDT 2017 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
robot@ev3dev:~$ lsusb
Bus 001 Device 002: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

So, as far as i can see, the ralink 7601 is recognized as hardware, but the networking layer is not operative

thanks for you help

dlech, 2018-04-22:
Fix released in `4.14.35-ev3dev-2.0.0-ev3` kernel.

closed 2018-04-22 by dlech