ev3dev/ev3dev #1037: ev3dev doesn't support an AC600 wifi dongle

cho934, 2018-02-11:
Hi,
For the next cup of robotic in March, we are allowed to use Wifi ac 5Ghz.

I've just tried 3 wiki dongles unsuccessfully :(

Netgear A6100
An AC600 USB adapter
TP-link AC450 nano USB adapter

Wifi is not available in brickman.
What can I do to solve quickly this problem ?

dlech, 2018-02-12:
See #1021

cho934, 2018-02-12:
A little bit difficult to understand :( and I don't want to compile everything...
It's working with a new kernel, isn't it ?

Is it possible to have a resume ?

dlech, 2018-02-12:
There are some Debian packages attached to that issue for specific kernel versions.

dlech, 2018-02-12:
Not sure what you mean by "resume"

cho934, 2018-02-12:
it means the final "how to" ?

dlech, 2018-02-12:
https://github.com/ev3dev/ev3dev/issues/1021#issuecomment-354867225

- download the zip file
- unzip it
- copy the .deb file to your EV3
- `dpkg -i <name-of-package>.deb

dlech, 2018-03-22:
Closing as duplicate of #1021