ev3dev/ev3dev #1038: OpenCV Python + EV3DEV + Webcam

Negociation, 2018-02-14:
Hi , i'm Marcos i from Brazil and i working on my final paper for college , anyway , i working on a Self-Driving Rover to use with AI ( Coffe / Googlenet ) , now i working on data capture , when i press a button from the IR control they get the data from sensors to create a instance. My problem now is get a photo from a usb webcam:

[image]

The Model of the webcam is the Logitech C920 works fine, i try to use fswebcam to take snapshots but its very slow , i checked on internet some people using OPENCV with Motion JPEG but i cant make works on my EV3 , I trying import but the terminal show me a message: no module named 'cv2':

My Dataset Generator Code:

`#!/usr/bin/env python3
from time import sleep
#from ev3dev.ev3 import *
import cv2
import threading

###################################################################
#		   DECLARACOES DE INICIALIZACAO			   #
###################################################################

#Dados de Revisao
Release = '2'
Version = '1'
Subversion = '3'

#Declaracao de Motores
Motor_Direito = LargeMotor('outB')
Motor_Esquerdo = LargeMotor('outC')

#Declaracao e Set do Sensor de Infravermelho para Controle Remoto
Infravermelho = InfraredSensor('in4')
Infravermelho.mode = 'IR-REMOTE'

#Declaracao do Sensor de Ultrassom para Distancia
Ultrasonic = UltrasonicSensor('in1')
Ultrasonic.mode = 'US-DIST-CM'

#Declaracao do Controle Remoto Esquerdo no Canal 1
ListenA = RemoteControl(channel=1)
ActionA = RemoteControl(channel=1)

#Declaracao do Controle Remoto Direito no Canal 4
ListenB = RemoteControl(channel=4)
ActionB = RemoteControl(channel=4)

Instancias = 0
Last_State = "none"

###########################################################################
#		   	     THREADS				          #
###########################################################################

#Thread Para Movimento com uso dos Controles Remotos
class Thread_Output (threading.Thread):
	def __init__(self):
		threading.Thread.__init__(self)
	def run(self):
		while (True):	
			ActionA.process()
			sleep(0.01)
			ActionB.process()

#Thread Para Set de Movimentos e Led
class Thread_Setting (threading.Thread):
	def __init__(self):
		threading.Thread.__init__(self)
	def run(self):
		while(True):	
			ListenA.process()
			sleep(0.01)
			ListenB.process()
			

#Thread Para Leitura dos Sensores com base nos movimentos Controle Remoto
class Thread_Input (threading.Thread):
	def __init__(self):
		threading.Thread.__init__(self)
	def run(self):
		global Last_State
		while (True):	
				if(( Last_State != "none") and (Last_State != "Stop" )):
					GenerateInstance(Last_State,GetDistancia(),GetLuminosidade())
					sleep(1)

###########################################################################
#		   	         EVENTOS LISTEN			         #
###########################################################################

#Funcao Referentes ao [ LISTEN ] do Controle Remoto A [ LEFT ]
def Listen_Left():
	def on_press(state):
		global Last_State
		if (state):
			Last_State = 'Left'
			Leds.all_off()
			Leds.set_color(Leds.LEFT,  Leds.GREEN)
	return on_press

#Funcao Referentes ao [ LISTEN ] do Controle Remoto A [ RIGHT ]
def Listen_Right():
	def on_press(state):
		global Last_State
		if (state):
			Last_State = 'Right'
			Leds.all_off()
			Leds.set_color(Leds.RIGHT,  Leds.GREEN)
	return on_press

#Funcao Referentes ao [ LISTEN ] do Controle Remoto B [ TOP ]
def Listen_Top():
	def on_press(state):
		global Last_State
		if (state):
			Last_State = "Foward"
			Leds.all_off()
			Leds.set_color(Leds.LEFT,  Leds.GREEN)
			Leds.set_color(Leds.RIGHT,  Leds.GREEN)
	return on_press

#Funcao Referentes ao [ LISTEN ] do Controle Remoto A [ DOWN ]
def Listen_Down():
	def on_press(state):
		global Last_State
		if (state):
			Last_State = "Stop"
			Leds.all_off()
			Leds.set_color(Leds.LEFT,  Leds.RED)
			Leds.set_color(Leds.RIGHT,  Leds.RED)
	return on_press

#Atribuicao de Funcoes ao Evento onPressed do Controle Remoto A [ LISTEN ]
ListenA.on_red_up = Listen_Left()
ListenA.on_red_down = Listen_Left()
ListenA.on_blue_up = Listen_Right()
ListenA.on_blue_down = Listen_Right()

#Atribuicao de Funcoes ao Evento onPressed do Controle Remoto B [ LISTEN ]
ListenB.on_red_up = Listen_Top()
ListenB.on_blue_up = Listen_Top()
ListenB.on_red_down = Listen_Down()
ListenB.on_blue_down = Listen_Down()

###########################################################################
#		   	         EVENTOS ACTION			         #
###########################################################################

#Funcao Referentes ao [ ACTION ] do Controle Remoto A [ LEFT ]
def Action_Left():
	def on_press(state):
		if (state):
			Validate('Left')
	return on_press

#Funcao Referentes ao [ ACTION ] do Controle Remoto A [ RIGHT ]
def Action_Right():
	def on_press(state):
		if (state):
			Validate('Right')
	return on_press

#Funcao Referentes ao [ ACTION ] do Controle Remoto B [ TOP ]
def Action_Top():
	def on_press(state):
		if (state):
			Validate('Foward')
	return on_press

#Funcao Referentes ao [ ACTION ] do Controle Remoto A [ DOWN ]
def Action_Down():
	def on_press(state):
		if (state):
			Validate('Stop') 
	return on_press

#Atribuicao de Funcoes ao Evento onPressed do Controle Remoto A [ ACTION ]
ActionA.on_red_up = Action_Left()
ActionA.on_red_down = Action_Left()
ActionA.on_blue_up = Action_Right()
ActionA.on_blue_down = Action_Right()

#Atribuicao de Funcoes ao Evento onPressed do Controle Remoto B [ ACTION ]
ActionB.on_red_up = Action_Top()
ActionB.on_blue_up = Action_Top()
ActionB.on_red_down = Action_Down()
ActionB.on_blue_down = Action_Down()

#Funcao de Validacao dos Movimentos
def Validate(Movimento):
	global StopRobot
	if Movimento=="Foward":
		Motor_Direito.run_forever(speed_sp=-300)
		Motor_Esquerdo.run_forever(speed_sp=-300)
	elif Movimento=="Stop":
		Motor_Direito.stop(stop_action="hold");
		Motor_Esquerdo.stop(stop_action="hold");
		GenerateInstance(Movimento,GetDistancia(),GetLuminosidade())	
	elif Movimento=="Left":
		Motor_Direito.run_forever(speed_sp=100)	
		Motor_Esquerdo.run_forever(speed_sp=-200)
	elif Movimento=="Right":
		Motor_Esquerdo.run_forever(speed_sp=100)	
		Motor_Direito.run_forever(speed_sp=-200)

#Obter a distancia do sensor de Ultrassom
def GetDistancia():
	DistanciaSensor = Ultrasonic.value()/10
	return DistanciaSensor

#Obter a Luminosidade do sensor de Cor
def GetLuminosidade():
	#Sensor_de_Cor.mode = 'COL-AMBIENT'
	#LuminosidadeSensor = Sensor_de_Cor.value()
	#return LuminosidadeSensor
	return 0

#Gerar Instancia para o Dataset 
def GenerateInstance(Movimento,Distancia,Luminosidade):
	global Instancias
	#Gerar Arquivo de Imagem
	os.system('fswebcam -d /dev/video0 -p MJPEG -r 176x144 --save /home/robot/%H%M%S.jpg')
	StringFile = "%s,%s,%s" % (Movimento,Distancia,Luminosidade)
	Instancias += 1
	print('Gerei um Dataset ( NOME_ARQUIVO.jpg - ' , StringFile ,')')
	

###################################################################
#		   INICIALIZACAO DO PROGRAMA			  #
###################################################################

#Mensagem de Boas Vindas
print('\n\n\n')
VersionData = Release + '.' + Version + '.' +Subversion
print('Inicializando o Gerador de Dados',VersionData)
print('----------------------------------------------------------')

#Bip de Inicializacao
Sound.beep()

#Inicializacao da Thread Input ( Leitura )
Input = Thread_Input()
Input.start()

#Inicializacao da Thread Set ( Led )
Setting = Thread_Setting()
Setting.start()

#Inicializacao da Thread Output ( Movimento )
Output = Thread_Output()
Output.start()

`

For Now my code still using fswebcam on the Method GenerateInstance , but i'll replace soon...  I look anothers code and they using python 2 i tried but they dont recognize the ev3dev import. I installed with apt-get opencv-python...

dlech, 2018-02-14:
ev3dev library is python3 only and opencv is python2 only. See #803

Negociation, 2018-02-14:
So what i can do , i need to use both at the same time....

dwalton76, 2018-02-14:
What I do is have a separate program that does everything involving OpenCV and call that program via subprocess.check_output() from my python3 ev3dev program.

dwalton76, 2018-02-14:
fswebcam is very slow for me too...it takes about 5s to take a picture :(

Negociation, 2018-02-14:
I can pass parameters to that program ? , i really dont know why ev3 library is not supported on Python 2 ... Basicily after a create a instance for my dataset i need to have a picture of what the robot see...

dwalton76, 2018-02-14:
You can pass parameters via the command line or if you have more complex parameters to pass write them to a file in json format and pass the name of that file as a parameter. 

Another option is to run your Python2/OpenCV program as a daemon and have your python3 program connect to it via a Unix domain socket.

Negociation, 2018-02-14:
You test uses OpenCV to take a picture ? it's faster enough , i using a delay of 1s for read data...

dwalton76, 2018-02-14:
I use fswebcam to take the pics and OpenCV to analyze them
http://programmablebrick.blogspot.com/2017/02/rubiks-cube-tracker-using-opencv.html?m=1

I haven’t tried taking pics via OpenCV but sounds like it would be worth trying if you are able to take a pic in 1s.

Negociation, 2018-02-14:
Its awesome your work , i'm newbie with everthing ( Linux ) , i just learned how do a apt-get few months ago ( and i'm a student of computer science ) ha-ha! . I'll try uses the subprocess first , i'll post a video of my work later to show , but i really wanna uses the EV3DEV with AI noone uses yet here ( Brazil ). If i understand the subprocess call a program who return a value ? But i need they works at the same time...

dwalton76, 2018-02-14:
Thanks :)  I am looking forward to your video.  What sort of AI will you be using?  I've read a few books on CNNs but I haven't done much in practice with them.

If the program you call via subprocess.check_output() exits 0 everything is fine, if it exits something other than 0 then check_output() will raise an exception.  Calling subprocess.check_output() can be slow though, especially since the program you will be calling will be importing the OpenCV library.  If you need this to be fast (it sounds like you do) then I would avoid using subprocess.check_output() and do the following instead:

- write a small python2 daemon (let's call it listener.py) that reads from a unix domain socket.  When it reads the string "TAKE A PICTURE" you have it snap a pic and then write some data back over the unix domain socket to let the client know the picture was taken (maybe send the filename of the pic to the client).  The first time you start this daemon it may take a few seconds to import OpenCV but after that it is just running in the background.

- your python3 code above (call it robot.py) will need to connect to listener.py and send the "TAKE A PICTURE" string and then wait for the response.  At that point robot.py can open and process the picture...or do you need OpenCV in order to process the picture?  If so you can have listener.py do the processing and return whatever results it finds, to robot.py via the unix domain socket (you would send this instead of sending the filename of the pic since robot.py wouldn't be doing anything with the pic since the pic was already processed listener.py).

Negociation, 2018-02-14:
A diagram from my application for help because all comments are in Portuguese : 
[image]

I will use two deep learning applications and compare the results: 
- Coffe AlexNet
- GoogleNET

1)Basicly i have a Procedure called GenerateInstance to create a TXT like 00001.txt where have the sensor values(Ultrassonic , Color ) and the move from Variable Latest_Move in file and i'll create a picture with the same name like 0001.jpg.

2) I have a Loop on Generate instance where i will create a new instance after 1s but i can increase the time in 0,5s (1,5s) except if i never pressed any button or i pressed to stop

3) I will press o button on my tablet ( WinForm application ) where i'll finish the capture session and after that transfer all files in a .zip like:

Dataset14-02-2018.zip:
001.txt
001.jpg
002.txt
002.jpg
003.txt
003.jpg
3) After that i will traning my neural network with the files of my dataset example in the instance 001 i have a picture of a wall when the ultrassonic sensor saw a object on 3,5cm...

This is just the capture for the pratice i wanna uses a webcam to watch the environment and take snapshots from frames sent to my neural network the picture and sensor values where my AI will return a move to do...

Negociation, 2018-02-14:
I probably wont save the picture on EV3 i guess ( I probably will streaming to my tablet using OpenCV (Motion JPEG) and save using My Tablet ( 300Mhz vs 1,8Ghz Quadcore Processor ).

Negociation, 2018-02-14:
I'll look that on friday with my Teacher who works with me on project for now i'll work in The Server Socket Application based on: https://github.com/lichiukenneth/EV3Dev-Python-Socket-Connection for Touchscreen using WinForms but looking like a UWP app with the basics like get the number of instances , shutdown Robot ...

xvzf, 2018-03-27:
https://pypi.python.org/pypi/opencv-python

OpenCV is indeed python 3.6 compatible, maybe give it another try! If you need help PM me

Negociation, 2018-03-27:
Hi , Xvzf tanks for your help , but i change my project , i'll use a Smartphone with Qpython to take the snapshots with OpenCV using a TCP Socket Connection to Sync , but i'll look for it...

Wolfyxon, 2024-12-17:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-opencv : Depends: libopencv-calib3d3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-features2d3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-highgui3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-imgcodecs3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-objdetect3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-stitching3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-superres3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-videoio3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-videostab3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: libopencv-viz3.2 (= 3.2.0+dfsg-4ev3dev1) but it is not going to be installed
                  Depends: python3-numpy (>= 1:1.10.0~b1) but it is not installable
                  Depends: python3-numpy-abi9 but it is not installable
                  Depends: libopencv-contrib3.2 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
I'm unable to install OpenCV

dlech, 2024-12-17:
Did you fix your `/etc/apt/sources.list` according to step 6.1 and do `sudo apt update`?

https://www.ev3dev.org/docs/getting-started/