ev3dev/ev3dev #1039: How to adjust font size in brickman?

dwalton76, 2018-02-21:
### What are you trying to do?
I have a raspberry pi3 with a 480x320 screen
https://www.amazon.com/gp/product/B01N447AEY/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

My gameplan is to add a brickpi3 to this setup but that isn't in the equation just yet.  The screen works but the font is super small and hard to read.  I need to adjust the font size but from some reading of previous issues it sounds like this is all hard coded.  The pistorms screen is 320x240 so I looked through the brickman source for references to 320 and 240 but do not see any.

Questions:
- where is the font size set for various platforms in brickman?
- How do I build/install brickman right on the pi3?  The README has instructions for how to build it via docker but it would be nice if I didn't have to mess with docker.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.9.59-ev3dev-1.6.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A21041
BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
BOARD_INFO_SERIAL_NUM=00000000dd20fe9a
BOARD_INFO_TYPE=main
```

dwalton76, 2018-02-21:
PS

There isn't an "Issues" tab at https://github.com/ev3dev/brickman so I assume we should open brickman issues here instead

dlech, 2018-02-21:
In ev3dev-stretch, BrickMan is DPI-aware. So, the ideal solution would be to patch the display driver to properly report the size of the screen.

As a workaround, you can use the `GRX_DRIVER` environment variable to tell it the correct DPI.

    GRX_DRIVER="dp 165"

Reference: http://docs.ev3dev.org/projects/grx/en/latest/c-api/mode.html#grx-set-driver

You should be able to set the for `brickman.service` by running `sudo systemctl edit brickman.service` and add

    [Service]
    Environment=GRX_DRIVER="dp 165"

dwalton76, 2018-02-23:
thanks, will give this a try

dwalton76, 2018-02-26:
If I do (roughly...am typing from memory here)
```
$ sudo su
$ export GRX_DRIVER="dp 165"
$ brickman
```
it works but no luck yet getting it to work when launching brickman via systemd

dwalton76, 2018-02-28:
woohoo got it working, the following is the trick (quotes are in different places)
```
robot@ev3dev[~]# cat /etc/systemd/system/brickman.service.d/override.conf
[Service]
Environment="GRX_DRIVER=dp 165"

robot@ev3dev[~]# 
```

systemd appears to be very picky about where you put the quotes

closed 2018-02-28 by dwalton76

dwalton76, 2018-11-01:
following up on this....dpi of 320 seems to be the magic spot for this screen.  The font is large enough to read easily at that setting.