ev3dev/ev3dev #1040: C++ motor working with OUTPUT_AUTO only

binarez, 2018-02-22:
### What are you trying to do?

Use multiple ev3 motors in C++. The following code works:

```
int main() {
    large_motor moteur(ev3dev::OUTPUT_AUTO);
    moteur.set_speed_sp(100);
    moteur.run_forever();
    return 0;
}
```

But it doesn't work when I specify ev3dev::OUTPUT_B to the motor ctor, it doesn't work even tho the motor really is connected to the B port. I know for a fact that the motor is connected to the B port because when I write out the return value of the address() method, it prints out "outB" when the motor is constructed using OUTPUT_AUTO.

### What did you expect to happen?

Being able to the specify explicitly which motor is bound to which variable in my code. I plan on using multiple motor so I cannot rely on AUTO.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-02-22:
Are you using the `jessie` branch of <https://github.com/ddemidov/ev3dev-lang-cpp>?

binarez, 2018-02-22:
I wasn't but it works when I do. Thanks a lot, closing this bug!

closed 2018-02-22 by binarez