ev3dev/ev3dev #1041: Cannot create regular file... Permission denied

CM1015, 2018-02-22:
### What are you trying to do?
I am trying to use
```
  root@ev3dev:~# cp /lib/systemd/system/bluetooth.service /etc/systemd/system/bluetooth.service
```
From this website: http://www.pabr.org/bricks/brickfpv/brickfpv.en.html 
to let a Bluetooth controller control my brick. 
### What did you expect to happen?
I expected the command to do what it needs to (I am not sure what exactly that is) and then I could move on to the next step:
"Using e.g. vi or nano, edit /etc/systemd/system/bluetooth.service to change the ExecStart line to:
	  ExecStart=/usr/lib/bluetooth/bluetoothd -P input

I am also not sure what that step means so if I could have some help with that as well that would be nice! 
### What actually happened?
```
cp: cannot create regular file '/etc/systemd/system/bluetooth.service': Permission denied
```

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-02-22:
Did you actually run this as root? e.g. (with `sudo`)

    sudo cp /lib/systemd/system/bluetooth.service /etc/systemd/system/bluetooth.service

dlech, 2018-02-22:
Of course, ev3dev has a kernel driver for PS3 controllers, so you shouldn't need to use the workaround from the post you linked anymore.

CM1015, 2018-02-22:
Yes I tried using sudo before and nothing happened. I don't need to use that anymore though?

dlech, 2018-02-22:
I guess that depends on what exactly you want to accomplish. There is also http://www.ev3dev.org/docs/tutorials/using-ps3-sixaxis/

CM1015, 2018-02-22:
I am trying to use a PS3 controller and webcam together. On the original website they used C to program, but what you sent me uses python so can I do that with Python?