ev3dev/ev3dev #1043: numpad for navigating the menu

dwalton76, 2018-03-02:
### What are you trying to do?

I am trying to use this numpad to navigate the brickman menu
https://www.amazon.com/dp/B01M4NH7F9/ref=sspa_dk_detail_8?psc=1&pd_rd_i=B01M4NH7F9&pd_rd_wg=hUfV8&pd_rd_r=MAMR49GRZHN41J6MAX48&pd_rd_w=hDjKB

The numpad is detected, it is the _China Resource Semico Co., Ltd_ entry
```
robot@ev3dev[~]# lsusb
Bus 001 Device 005: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 006: ID 1a2c:0e24 China Resource Semico Co., Ltd 
Bus 001 Device 004: ID 046d:0826 Logitech, Inc. HD Webcam C525
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
robot@ev3dev[~]# 
```
and dmesg shows that it was detected as a keyboard
```
[   29.866066] usb 1-1.4: new low-speed USB device number 6 using dwc_otg
[   30.009556] usb 1-1.4: New USB device found, idVendor=1a2c, idProduct=0e24
[   30.009585] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   30.009603] usb 1-1.4: Product: USB Keyboard
[   30.009618] usb 1-1.4: Manufacturer: SEM
[   30.027927] input: SEM USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:1A2C:0E24.0001/input/input1
[   30.097573] hid-generic 0003:1A2C:0E24.0001: input,hidraw0: USB HID v1.10 Keyboard [SEM USB Keyboard] on usb-3f980000.usb-1.4/input0
[   30.113358] input: SEM USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1A2C:0E24.0002/input/input2
[   30.176657] hid-generic 0003:1A2C:0E24.0002: input,hidraw1: USB HID v1.10 Device [SEM USB Keyboard] on usb-3f980000.usb-1.4/input1
[   37.529943] random: crng init done
````

### What actually happened?
`Back Space` works but nothing else (I have tried toggling numlock)

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.9.59-ev3dev-1.6.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A21041
BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
BOARD_INFO_SERIAL_NUM=00000000dd20fe9a
BOARD_INFO_TYPE=main
```

dlech, 2018-03-02:
Can you show the output of `evtest` when pressing the arrow keys on the keypad (and the working backspace for good measure).

dwalton76, 2018-03-02:
evtest sees two keyboards for some reason, not sure if this is a factor or not.  I select keyboard 0

```
robot@ev3dev:~$ evtest 
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	SEM USB Keyboard
/dev/input/event1:	SEM USB Keyboard
Select the device event number [0-1]: 0 
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1a2c product 0xe24 version 0x110
Input device name: "SEM USB Keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
[... 180 lines cut ...]
Event: time 1520016759.329216, type 1 (EV_KEY), code 75 (KEY_KP4), value 0
Event: time 1520016759.329216, -------------- SYN_REPORT ------------

Event: time 1520016760.985220, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7005e
Event: time 1520016760.985220, type 1 (EV_KEY), code 77 (KEY_KP6), value 1
Event: time 1520016760.985220, -------------- SYN_REPORT ------------
Event: time 1520016761.089225, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7005e
Event: time 1520016761.089225, type 1 (EV_KEY), code 77 (KEY_KP6), value 0
Event: time 1520016761.089225, -------------- SYN_REPORT ------------

Event: time 1520016762.561234, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7005a
Event: time 1520016762.561234, type 1 (EV_KEY), code 80 (KEY_KP2), value 1
Event: time 1520016762.561234, -------------- SYN_REPORT ------------
Event: time 1520016762.641221, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7005a
Event: time 1520016762.641221, type 1 (EV_KEY), code 80 (KEY_KP2), value 0
Event: time 1520016762.641221, -------------- SYN_REPORT ------------

Event: time 1520016764.241224, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70058
Event: time 1520016764.241224, type 1 (EV_KEY), code 96 (KEY_KPENTER), value 1
Event: time 1520016764.241224, -------------- SYN_REPORT ------------
Event: time 1520016764.313209, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70058
Event: time 1520016764.313209, type 1 (EV_KEY), code 96 (KEY_KPENTER), value 0
Event: time 1520016764.313209, -------------- SYN_REPORT ------------

```

dlech, 2018-03-02:
Is there any output for the other keyboard device?

Is there any output when you press the num lock key?

does the num lock LED work?

dlech, 2018-03-03:
The keypad I have actually sends out the expected key codes, e.g.

```
Event: time 1520097679.872227, -------------- SYN_REPORT ------------
Event: time 1520097680.072227, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7004e
Event: time 1520097680.072227, type 1 (EV_KEY), code 109 (KEY_PAGEDOWN), value 1
```

Although it does not send out a numlock key code. Brickman uses libinput for keyboard/mouse input (from Wayland display manager project), so I would expect it to "just work". But, maybe there is something special that needs to be done to handle the numlock state.

dlech, 2018-03-03:
Does this keypad work as expected with other programs?

dwalton76, 2018-03-03:
yeah I tried it in a text editor on my laptop (ubuntu 17.10) and it works just fine

dwalton76, 2018-03-03:
I read some stuff where ubuntu has an option to use a numpad to control the mouse...maybe we are getting tripped up on something like that?