ev3dev/ev3dev #1044: BOARD_INFO_MODEL for brickpi3

dwalton76, 2018-03-02:
### What are you trying to do?

The BOARD_INFO_MODEL for a brickpi3 shows up as `Raspberry Pi 2 Model B Rev 1.1`
```
robot@ev3dev[~]# cat  /sys/class/board-info/board0/uevent 
BOARD_INFO_HW_REV=A21041
BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
BOARD_INFO_SERIAL_NUM=00000000dd20fe9a
BOARD_INFO_TYPE=main
robot@ev3dev[~]# 
```
but [get_current_platform() ](https://github.com/ev3dev/ev3dev-lang-python/blob/develop/ev3dev2/__init__.py#L78-L80) in ev3dev-lang-python is looking for `Dexter Industries BrickPi3`.  I am not sure if it ever displayed `Dexter Industries BrickPi3` or if we just assumed that would be the output based on the output for the original brickpi.

Could we have BOARD_INFO_MODEL display `Dexter Industries BrickPi3` here?

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.9.59-ev3dev-1.6.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A21041
BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
BOARD_INFO_SERIAL_NUM=00000000dd20fe9a
BOARD_INFO_TYPE=main
```

dlech, 2018-03-02:
There should be a second board entry for the BrickPi3. Did you configure it in `config.txt`?

dwalton76, 2018-03-02:
I had but now that I am double checking I find that my `config.txt` is empty...digging

dwalton76, 2018-03-02:
restoring config.txt (with the brickpi3 line uncommented) fixed it

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.9.59-ev3dev-1.6.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

closed 2018-03-02 by dwalton76