ev3dev/ev3dev #1046: nxt-python

Vienhackar, 2018-03-13:
I am trying to allow the ev3 to master over the nxt, I found this [post](https://bricks.stackexchange.com/questions/4512/is-it-possible-to-connect-ev3-and-nxt-bricks-like-master-slave-configuration-via) that says to install nxt-python on ev3dev, but I have no idea how to do this, I have no coding expirience, I am just a simple man looking for an easy solution.

dlech, 2018-03-13:
Make sure you EV3 is connected to the internet, then run

    sudo apt update
    sudo apt install python-nxt

closed 2018-03-16 by Vienhackar

bonmotwang, 2018-04-08:
Thanks. I just had my son started on EV3 using Python. Found a great deal on a used full set of NXT. This is a great news that he can talk to NXT and expand the inputs and outputs. 
I have another question. Does EV3DEV works on NXT? I am thinking about using NXT as a co-processor to share some workload from EV3. 
Thanks.

bonmotwang, 2018-04-08:
Does “remote control” reads sensors connected to the NXT? Thanks

dlech, 2018-04-09:
https://github.com/Eelviny/nxt-python/wiki

bonmotwang, 2018-04-09:
Wow. Amazing. Reading sensors! Running motors! Even display message on NXT!
You guys are awesome!