ev3dev/ev3dev #1047: Programming ev3dev

Vienhackar, 2018-03-14:
I am completely new to ev3dev, and I have just a little coding experience but not enough, can I still program ev3dev with the standard ev3 programming software? or something similar?

dlech, 2018-03-14:
You *can* use the EV3-G programming software with ev3dev, but it requires some special setup. See <https://github.com/ev3dev/lms2012-compat>.

Most people using ev3dev use python. See <https://sites.google.com/site/ev3python/> for a guide.

Vienhackar, 2018-03-14:
Okay, so I tried to follow the instructions but when I type sudo lms2012 it said unknown command

dlech, 2018-03-14:
for lms2012 yes, you need ev3dev-stretch

Vienhackar, 2018-03-14:
So I need to flash it to the sd card and get rid of jessie?

WasabiFan, 2018-03-15:
If you are using ev3dev, I would expect you are looking to be able to use its special abilities rather than simply emulating the standard EV3 software. If that expectation is correct, you should get started with Python. Otherwise, _why_ are you trying to run the normal programming language on the ev3dev platform? What do you get over just using the stock firmware?

dlech, 2018-03-15:
if you want to use lms2012, then yes. However, it doesn't really do anything that the official LEGO firmware does, so it is probably not very useful unless you have some other motivation for using it.

Vienhackar, 2018-03-15:
I was told by installing the nxt-package I could us it to communicate to the nxt, Basically I have nxt and ev3, and I want one to control the other, I would have simply used the nxt to control the ev3 however I need it to use ev3's gyro sensors
so I just want the ev3 to be able to master over the nxt nothing fancy

Vienhackar, 2018-03-15:
Would I do this the same as download the normal ev3dev?

closed 2018-03-16 by Vienhackar