ev3dev/ev3dev #1049: Using the Mindsensor NumericPad

jlehuen, 2018-03-15:
Hello,

I have to get the value of the [Mindsensor NumericPad](http://www.mindsensors.com/ev3-and-nxt/66-numericpad) but this sensor is not taken into account the ev3dev libraries.

How can I use this device with C langage (ev3dev-c) ?

Thank you
Jerome

dlech, 2018-03-15:
You will need to use I2C commands directly using the information in the Users Guide from mindsensors. It will have a section on I2C registers.

Here are some resources:

* http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html
* https://www.kernel.org/doc/Documentation/i2c/dev-interface

jlehuen, 2018-03-15:
Have you got a starting example with ev3dev-c for an existing sensor?
It would help me get started...
Thanks

dlech, 2018-03-16:
For this sensor, ev3dev-c will not help you at all. The second link I gave has an example in C of how to write i2c messages, like which header files you need to include.

There is a python example [here](https://github.com/ev3dev/ev3dev/issues/1007#issuecomment-344684487). The function names are basically the same, so it should be easy to translate, e.g. `n = bus4.read_byte_data(0x42)` would be `s32 n = i2c_smbus_read_byte_data(bus4, 0x42)` in C.

dlech, 2018-03-16:
The first link I gave will tell you why you need the address 0x5a instead of 0xb4 like it says in the mindsenors.com documentation and why we use `input_port + 2`.

No idea if this works, but maybe something like this...

```C
#include <linux/i2c-dev.h>
#include <i2c/smbus.h>

#define BIT(n) (1 << (n))

const int input_port = 1; /* the input port */
const int addr = 0x5a; /* The I2C address */

int file;
int adapter_nr = input_port + 2;
char filename[20];
s32 keys;
  
snprintf(filename, 19, "/dev/i2c-%d", adapter_nr);
file = open(filename, O_RDWR);
if (file < 0) {
    /* ERROR HANDLING; you can check errno to see what went wrong */
    exit(1);
}

if (ioctl(file, I2C_SLAVE, addr) < 0) {
    /* ERROR HANDLING; you can check errno to see what went wrong */
    exit(1);
}

keys = i2c_smbus_read_word_data(file, 0x00);
if (keys < 0) {
    /* ERROR HANDLING; you can check keys to see what went wrong */
    exit(1);
}

if (keys & BIT(0)) {
    /* # pressed */
}

if (keys & BIT(1)) {
    /* 9 pressed */
}

/* and so on */

if (keys & BIT(11)) {
    /* 4 pressed */
}
```

I might have the byte order wrong where `BIT(0)` is * and `BIT(8)` is # instead.

jlehuen, 2018-03-16:
Thank you very much, I'm going to try that.
Jerome.

jlehuen, 2018-04-05:
Hi David,

I tried to adapt the method which is implemented in the file "NP-lib-original.nxc" from Mindsensor but without any success. Do you have an idea why my driver " keypad.c" does not work ?

$ gcc -c smbus.c 
$ gcc -c keypad.c 
$ gcc smbus.o keypad.c -o keypad

Thanks,

Jerome Lehuen
Le Mans University
France

jlehuen, 2018-04-05:
In the file below I changed the hexa codes (from the original file) but I don’t understand their signification (no documentation from MindSensor) and I change the second parameters of the i2c_smbus_write_block_data function but I don’t really understand it also.

Jerome

dlech, 2018-04-05:
GitHub doesn't do email attachments. You will need to include the code inline or via the web interface so that others can see it.

jlehuen, 2018-04-06:
Hello,

I think I found a solution based on the RobotC code (so the problem seems solved). That said, I still don't understand the hexa configuration code.

~~~c
#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <sys/types.h>
#include <linux/i2c-dev.h>
#include "smbus.h"

#define Wait( msec ) usleep(( msec ) * 1000 )
typedef enum { FALSE, TRUE } bool;
void _beep() { system("beep -f 800 -l 50"); }

#define I2C_ADDR 0x5A // Unshifted 7 bits address (do not use 0xB4)
#define DATA_REG 0x00 // Data register

// ----------------------------------------------------------------
// Open the I2C port
// ----------------------------------------------------------------

int open_i2c_dev(int port) {
	char filename[20];
	snprintf(filename, 19, "/dev/i2c-%d", port+2);
	int file = open(filename, O_RDWR);
	if (file < 0) {
		fprintf(stderr, "I2C error: Could not open file %s\n", filename);
		exit(1);
	}
	if (ioctl(file, I2C_SLAVE, I2C_ADDR) < 0) {
		fprintf(stderr, "I2C error: Failed to set slave address 0x%x\n", I2C_ADDR);
		exit(1);
	}
	return file;
}

// ----------------------------------------------------------------
// Initialize the keypad
// ----------------------------------------------------------------

int KP_initialized[] = { FALSE, FALSE, FALSE, FALSE };

const __u8 KP_config_group1[] = { 0x2B, 0x01, 0x01, 0x00, 0x00, 0x01, 0x01, 0xFF, 0x02 };
const __u8 KP_config_group2[] = { 0x41, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A };
const __u8 KP_config_group3[] = { 0x4F, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A, 0x0F, 0x0A };
const __u8 KP_config_group4[] = { 0x5C, 0x0B, 0x20, 0x0C };
const __u8 KP_config_group5[] = { 0x7B, 0x0B };
const __u8 KP_config_group6[] = { 0x7D, 0x9C, 0x65, 0x8C };

bool KP_Init(int port) {

	int file = open_i2c_dev(port);
	KP_initialized[port] = file;

	if (i2c_smbus_write_block_data(file, I2C_ADDR, sizeof(KP_config_group1), KP_config_group1) < 0) return FALSE;
	if (i2c_smbus_write_block_data(file, I2C_ADDR, sizeof(KP_config_group2), KP_config_group2) < 0) return FALSE;
	if (i2c_smbus_write_block_data(file, I2C_ADDR, sizeof(KP_config_group3), KP_config_group3) < 0) return FALSE;
	if (i2c_smbus_write_block_data(file, I2C_ADDR, sizeof(KP_config_group4), KP_config_group4) < 0) return FALSE;
	if (i2c_smbus_write_block_data(file, I2C_ADDR, sizeof(KP_config_group5), KP_config_group5) < 0) return FALSE;
	if (i2c_smbus_write_block_data(file, I2C_ADDR, sizeof(KP_config_group6), KP_config_group6) < 0) return FALSE;

	Wait(100);
	return TRUE;
}

void KP_Close() {
	int port;
	for (port = 1 ; port < 5 ; port++)
		if (KP_initialized[port])	
			close(KP_initialized[port]);
}

// ----------------------------------------------------------------
// Read a binary value from the keypad
// ----------------------------------------------------------------

int KP_Read(int port) {

	if (!KP_initialized[port] && !KP_Init(port)) {
		fprintf(stderr, "I2C error: Failed to init device at port %d\n", port);
		exit(1);
	}
	int file = KP_initialized[port];
	int result = -1;
	int size = 2;
	__u8 data[2];

	if (i2c_smbus_read_i2c_block_data(file, DATA_REG, size, data) != size) {
		fprintf(stderr, "I2C error: Failed to read block at port %d\n", port);
		exit(1);
	}
	result = data[0] + data[1]*256;
	result = result & 0x0FFF; // The last 4 bits of byte 2 are not relevant
	Wait(50);
	return result;
}

// ----------------------------------------------------------------
// Convert the binary value into an integer code
// ----------------------------------------------------------------

const int KP_keyMap[] = { 11, 9, 6, 3, 0, 8, 2, 5, 10, 7, 1, 4 };

int ReadKeypad(int port) {
	
	int i, keyPress = KP_Read(port);
	for (i = 0 ; i < 12 ; i++) {
		if (keyPress & (1 << i)) {
			return KP_keyMap[i];
		}
	}
	return -1;
}

// ----------------------------------------------------------------
// Main program
// ----------------------------------------------------------------

void main() {

	int key = -1;
	printf("Start scanning...\n");
	while (key != 11) {
		key = ReadKeypad(1);
		while (key < 0) key = ReadKeypad(1);
		_beep();
		printf("%d\n", key);
		while (key == ReadKeypad(1));
	}
	KP_Close();
}
~~~

jlehuen, 2018-04-06:
The problem seems solved.

closed 2018-04-06 by jlehuen