ev3dev/ev3dev #1050: lms2012 no video driver

Vienhackar, 2018-03-15:
So I was trying to get lms2012 to run, I downloaded stretch and installed lms2012 and ran sudo lms2012
_______________________________________________________________________________________________________________
robot@ev3dev:~$ sudo lms2012
[sudo] password for robot:

** (process:925): WARNING **: Could not start graphics: No supported video driver plugins found

** (process:925): WARNING **: Falling back to memory driver

** (process:925): ERROR **: Could not load memory driver
Trace/breakpoint trap
**robot@ev3dev:~$**
_______________________________________________________________________________________________________________

dlech, 2018-03-15:
I believe this has been fixed already. Upgrade the `lms2012-compat` package.

* make sure EV3 is connected to the internet
* in an SSH terminal, run `sudo apt update && sudo apt install lms2012-compat`

Vienhackar, 2018-03-16:
sorry about that, forgot I posted that, new to github, I tried that and got this:
** (process:3292): WARNING **: Failed to open /dev/hidg0: No such file or directory

** (process:3292): WARNING **: Error setting property 'Discoverable' on interface org.bluez.Adapter1: GDBus.Error:org.bluez.Error.Failed: Not Powered (g-io-error-quark, 36)

dlech, 2018-04-15:
Closing since original issue is solved. I'm also looking at packaging a python3 version of nxt-python.

closed 2018-04-15 by dlech