ev3dev/ev3dev #1052: Does exist a list of differences between EV3Dev jessie and EVEDev stretch?

jabrena, 2018-03-18:
Hi,

I would like to know the list of changes between EV3Dev Jessie and EVEDev Stretch.
I would like to open a Dev branch in the `Java` library to add support for Stretch distro.

Any address with a summary with the changes?

Cheers

Juan Antonio

dlech, 2018-03-19:
You can find some hints in the kernel changelogs. e.g. https://github.com/ev3dev/ev3dev-kpkg/blob/ev3dev-stretch/ev3dev-ev3/changelog

tcwan, 2018-03-20:
Well, for one, all the sysfs paths for the devices have changed. May not affect you unless you're working on libraries that have not been updated.

dlech, 2018-03-20:
> Well, for one, all the sysfs paths for the devices have changed.

This should not be the case. The value of the `address` attribute has changed for all devices. Perhaps that is what you are referring tos?

tcwan, 2018-03-21:
For example, in Jessie:
"/sys/class/leds/ev3:left:red:ev3dev/brightness"
became in Stretch:
"/sys/class/leds/led0:red:brick-status/brightness"

I haven't tried going through all the devices yet, but this is the first that I've noticed.

jabrena, 2018-03-25:
The port names for EV3 Has changed too.
Is it correct @dlech ?

``` 
robot@ev3dev:/sys/class/lego-port/port0$ ls
address  device  driver_name  ev3-ports:in1:ev3-uart-host  mode  modes  power  set_device  status  subsystem  uevent
robot@ev3dev:/sys/class/lego-port/port0$ cat address 
ev3-ports:in1
```

jabrena, 2018-03-25:
What is the purpose of `decimals`, `bin_data`, `bin_data_format`?

```
robot@ev3dev:/sys/class/lego-sensor/sensor5$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat driver_name 
```

jabrena, 2018-03-30:
Another difference is the path for battery:

**Jessie:**
/sys/class/power_supply/legoev3-battery

**Stretch:**
/sys/class/power_supply/lego-ev3-battery

jabrena, 2018-03-31:
Another difference is the EV3 Key path:

Jessie:
/dev/input/by-path/platform-gpio-keys.0-event 

Stretch:
??

@dlech, @tcwan what is the new address for EV3 Keys?

jabrena, 2018-03-31:
Another difference with EV3 Speaker:

How to set the volume?

**Jessie:**
/sys/devices/platform/snd-legoev3/

Examples to use the path by command line:
http://www.ev3dev.org/docs/tutorials/using-ev3-speaker/

**Stretch:**
??

How to set the volume?
How to read the volume?

@dlech , @tcwan How to fix it?

dlech, 2018-03-31:
`amixer` or use libalsa directly.

tcwan, 2018-04-16:
I'm curious why `stretch` has such convoluted sysfs paths for devices:
```
Debian Jessie
=============
/sys/class/tacho-motor/motor0 -> ../../devices/platform/legoev3-ports/lego-port/port6/outC:lego-ev3-l-motor/tacho-motor/motor0
/sys/class/tacho-motor/motor1 -> ../../devices/platform/legoev3-ports/lego-port/port5/outB:lego-ev3-l-motor/tacho-motor/motor1

Debian Stretch
==============
/sys/class/tacho-motor/motor0 -> ../../devices/platform/ev3-ports/ev3-ports:outC/lego-port/port6/ev3-ports:outC:lego-ev3-l-motor/tacho-motor/motor0
/sys/class/tacho-motor/motor1 -> ../../devices/platform/ev3-ports/ev3-ports:outB/lego-port/port5/ev3-ports:outB:lego-ev3-l-motor/tacho-motor/motor1
```

In `jessie`, the path is pretty consistent and follows a platform to physical to logical progression. However, in `stretch`, `ev3-ports:outC` appears twice in the path, and it goes from a (platform?) logical description to a physical description and then back to a logical description again. While this wouldn't affect access to the actual device if we go through `/sys/class/tach-motor/...`, it makes parsing the symbolic path complex if there ever is a need to enumerate the ports to detect attached devices. For example, I can't do a basic string search for `outX` since it appears twice in the path.

dlech, 2018-04-16:
That's just how the kernel works. It is a graph of kernel objects. If you want to enumerate devices, I highly recommend using libudev.

tcwan, 2018-04-16:
> That's just how the kernel works. It is a graph of kernel objects. If you want to enumerate devices, I highly recommend using libudev.

Ha. This looks like it'll be another rabbit-warren to get lost in. Thanks for the info though, I didn't know about `libudev` before.

jabrena, 2018-04-16:
hehehe, @dlech you enjoy with Linux so much, but later the rest of tribe has to dance with the rhythm :p 

I will review how Python library solve the problem and I would try to solve.
One question. Using GRX, is not another way to solve Display features?

Juan Antonio

jabrena, 2018-04-22:
https://github.com/ev3dev/ev3dev-lang-python/pull/460