ev3dev/ev3dev #1053: The "power" checkbox is not there at the latest version of ev3dev

coolshare, 2018-03-20:
### What are you trying to do?
install ev3dev

### What did you expect to happen?
detect my WI-FI dougle
And some old instructions said to check the "power" checkbox but I could not find it in the latest version
### What actually happened?
It is not detecting none

### What hardware and software are you using (including version numbers)?

ev3dev at
https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2017-09-14/ev3dev-jessie-ev3-generic-2017-09-14.zip

used EV3 31313

dlech, 2018-03-20:
From Gitter:

> my WI-FI dougle (WNDA3100)

dlech, 2018-03-20:
I'm guessing that we don't have a driver for this dongle. Please provide the output of the `lsusb` command so we know exactly which dongle you have. It looks like there are 3 versions of WNDA3100.

https://wikidevi.com/wiki/Netgear_WNDA3100v1