ev3dev/ev3dev #1054: bin_data does not provide useful values when using the EV3 touch sensor

QuirkyCort, 2018-03-20:
### What are you trying to do?

Read from bin_data when using the EV3 touch sensor.

### What did you expect to happen?

Receive either a value representing the state (0 or 1) or a value representing the analog reading.

### What actually happened?

bin_data provided only the lower 8 bits of the 16 bits analog reading. This makes the value essential worthless, as we cannot use it to reliably determine the state of the touch sensor.

I'll readily admit that there are very little (...if any) reasons to read from bin_data, as the sensor state could be determined by reading from value0 instead. But if bin_data is going to provide a readable value, it'll make more sense to ensure that the value is useful by providing the full 16 bits.

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

tcwan, 2018-04-16:
Is this commit part of `snapshot-ev3dev-stretch-ev3-generic-2018-04-14.img.xz`?

dlech, 2018-04-16:
no. the `fix committed` tag only means that a git commit exists. I will update the issue again when the fix has been released.

dlech, 2018-04-22:
Fix released in `ev3dev-2.0.0` kernels.

closed 2018-04-22 by dlech