ev3dev/ev3dev #1055: Writing to LED brightness for the EV3 onboard LED causes system slowdown

QuirkyCort, 2018-03-20:
### What are you trying to do?

Set the LED to either on or off.

When programming a simple line following robot using Python, I set the EV3's onboard LED color according to the action being taken by the robot (ie. Left LED on when turning left, Right LED on when turning right, both on when going straight). I noticed that the program performance was very poor, running as low as 4 loops per second on the main control loop. Furthermore, I observed that the poor performance continue for several minutes even after the LED commands were removed from the program. It was found using top that system load (sy, time running kernel process) was high for several minutes after the program ended.

### What did you expect to happen?

Setting the onboard LED brightness should not cause high system load.

### What actually happened?

Further test on the command line shows repeated writing to brightness will cause high system load, and that this can persist for several minutes after the write completes, depending on the number of times the write was done. A sample bash script demonstrates this problem:

```
for in in $(seq 1 100); do 
 echo 0 > /sys/class/leds/ev3\:left\:green\:ev3dev/brightness
 echo 0 > /sys/class/leds/ev3\:left\:red\:ev3dev/brightness
 echo 0 > /sys/class/leds/ev3\:right\:green\:ev3dev/brightness
 echo 0 > /sys/class/leds/ev3\:left\:red\:ev3dev/brightness
done

while true; do 
 top -b -n2 -d1 | grep %Cpu |tail -n1
 date
done # High system load was observed for about 1 minute
```

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2018-03-22:
does this also happen on 4.9.x kernels (ev3dev-stretch)?

QuirkyCort, 2018-03-23:
Just tested it with the latest stretch snapshot (2018-03-16). It's a little harder to tell, as in stretch, the system load is constantly high (probably due to the ADC as mentioned in https://github.com/ev3dev/ev3dev/issues/1022).

I ran a simple Python benchmark before and after the LED brightness writes, and it seems that stretch isn't affected. The results are as follows:

**Jessie (Before brightness write)**
Avg 1.1 seconds to complete

**Jessie (After brightness write)**
Avg 3.6 seconds to complete

**Stretch (Same results before and after brightness write)**
Avg 1.4 seconds to complete

Overall, stretch is slower than jessie, but it's performance isn't affected by the LED.

dlech, 2018-04-15:
ev3dev-jessie is no longer being developed, so I'm closing this issue since it seems to be fixed in ev3dev-stretch.

closed 2018-04-15 by dlech