ev3dev/ev3dev #1056: debian-jessie-cross docker image is broken

dlech, 2018-03-22:
Source file: https://github.com/ev3dev/docker-cross/blob/master/ev3dev-jessie/debian-jessie-cross.dockerfile

See <https://lists.debian.org/debian-embedded/2018/03/msg00000.html> for explanation of the problem.

jabrena, 2018-03-22:
Hi @dlech, many thanks for the link.

Last week, I tried to execute the Branch to release `OpenJDK 10:`
https://github.com/ev3dev-lang-java/openjdk-ev3/tree/openjdk-10

and I couldn´t for the reason that you expose. 
Do you know other alternative for Jessie?

Related issue:
https://github.com/ev3dev-lang-java/openjdk-ev3/issues/7

Cheers

Juan Antonio

dlech, 2018-03-22:
> Do you know other alternative for Jessie?

Stretch?

dlech, 2018-03-22:
No, I just meant that we should forget about jessie and focus on stretch for new development. Since jessie is old-stable, I don't expect that this issue will be fixed.

jabrena, 2018-03-22:
For me, it is Ok. 

This weekend, I will review in detail the changes in the ev3dev /sys/class and I could create a quick new version with the changes. But I need to solve the OpenJDK build process. I will work this weekend in this area with your dvice about Stretch.

Many thanks @dlech

jabrena, 2018-03-22:
In Java side for EV3, I need to have a stable solution to build Java. Now appear a new version every 6 month.

reznikmm, 2018-04-21:
Trying to build docker image from debian-jessie-cross.dockerfile I got this:
```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 crossbuild-essential-armel : Depends: gcc-arm-linux-gnueabi (>= 4.9.1-1) but it is not going to be installed
                              Depends: g++-arm-linux-gnueabi (>= 4.9.1-1) but it is not going to be installed
 crossbuild-essential-armhf : Depends: gcc-arm-linux-gnueabihf (>= 4.9.1-1) but it is not going to be installed
                              Depends: g++-arm-linux-gnueabihf (>= 4.9.1-1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
Any ideas how to fix?

dlech, 2018-04-21:
You could try using the `8.9` tag.

    FROM debian:8.9

jabrena, 2018-05-04:
Hi @dlech,

we tried but it continue failing:
https://github.com/ev3dev-lang-java/openjdk-ev3/issues/23

Juan Antonio

dlech, 2018-07-20:
I doubt the emdebian repo will ever be updated, so marking this wontfix.

The only workaround it to base your own dockerfile off of [the ev3dev/debian-jessie-cross image on docker hub](https://hub.docker.com/r/ev3dev/debian-jessie-cross/) instead of trying to build or modify `debian-jessie-cross.dockerfile` yourself.

closed 2018-07-20 by dlech