ev3dev/ev3dev #1057: EV3Dev support for RaspberryPi 3 B+

jabrena, 2018-03-24:
Hi @dlech,

One user from the Java community, @woolfel, has reported that EV3Dev jessie image doesn´t  run on the new Raspberry 3 B+. The test uses the following combination:

Image: Debian Jessie
Board: BrickPi 2.8
RaspberryPi: 3B+

Can you review please?

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/432#issuecomment-375920724

Juan Antonio

woolfel, 2018-03-24:
Correction. I tested the current image on the older Pi 3 B and BrickPi 2.8

I did try the default Jessie image on Pi 3 B+ and it wouldn't even boot, so that's a Raspbian issue with 3 B+ :)

dlech, 2018-03-24:
Sorry, we won't be fixing this for ev3dev-jessie (the kernel is too old). We will add support to ev3dev-stretch though.

jabrena, 2018-03-24:
Many thanks @dlech

woolfel, 2018-03-24:
no problem. Just letting you know the image had an error :)

dlech, 2018-04-09:
https://github.com/DexterInd/BrickPi3/pull/90

mattallen37, 2018-04-09:
FYI DexterInd/BrickPi3#90 relates to BrickPi3 only. It doesn't apply to BP prior to 3.x (such as 2.8).

dlech, 2018-04-09:
@mattallen37, ACK. We've packaged up your script in a debian package, so we need to update it.

https://github.com/ev3dev/firmware-brickpi3

I just put it here since this issue is a catch-all for RPi3 B+ compatibility issues.

dlech, 2018-04-11:
The main missing part is the kernel. However, in the mean time, the RPi has moved to a 4.14 kernel, so we will have to update the ev3dev kernel to 4.14 as well. Maybe in a couple weeks if I don't get distracted with something else.

zombiezen, 2018-04-11:
Wonderful, thanks @dlech!

dlech, 2018-04-14:
Please try the `2018-04-14` snapshot image.

https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/

zombiezen, 2018-04-14:
Yay, this got much farther on mine!  It loads all the way up to Brickman, stays there for a bit, and then appears to do a software shutdown (judging based on the logs that appear on my plugged in monitor).  Is there any way I can pull logs to give you more debugging information?

dlech, 2018-04-14:
>Do I always need to have the battery hooked up to the BrickPi, even if I'm not running any motors?

See #989.

zombiezen, 2018-04-15:
Got it. Only other thing: seems like it can't find the Wi-Fi hardware:

```
robot@ev3dev:~$ connmanctl
connmanctl> enable wifi
Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist
```

dlech, 2018-04-15:
I haven't looked yet. Does the 3 B+ have a new/different Wi-Fi chip?

dlech, 2018-04-15:
I'm guessing yes because there is a new firmware package released on pi day. I've imported that into the ev3dev repo now.

Try...

```
sudo apt update
sudo apt upgrade
sudo reboot
```
 
and see what happens.

dlech, 2018-04-15:
Meh.. I spoke too soon. Looks like it imported an older version for some reason... Trying again...

dlech, 2018-04-15:
ok, that's straightened out now, back to https://github.com/ev3dev/ev3dev/issues/1057#issuecomment-381374169

zombiezen, 2018-04-15:
I kicked off an apt upgrade and now I'm getting an error quite similar to the one in #386:

```
Processing triggers for initramfs-tools (0.130) ...
update-initramfs: Generating /boot/initrd.img-4.14.33-ev3dev-1.6.1-rpi2
Unsupported platform.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

(Previous output if it helps any:)

```
Get:1 http://security.debian.org stretch/updates InRelease [94.3 kB]
Get:2 http://archive.ev3dev.org/debian stretch InRelease [8,416 B]
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Get:4 http://cdn-fastly.deb.debian.org/debian stretch Release [118 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian stretch Release.gpg [2,434 B]
Get:6 http://archive.ev3dev.org/debian stretch/main armhf Packages [49.2 kB]
Get:7 http://cdn-fastly.deb.debian.org/debian stretch/main armhf Packages [6,950 kB]
Get:8 http://cdn-fastly.deb.debian.org/debian stretch/contrib armhf Packages [42.0 kB]
Get:9 http://cdn-fastly.deb.debian.org/debian stretch/non-free armhf Packages [59.6 kB]
Fetched 7,324 kB in 16s (457 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bluez-firmware brickd firmware-atheros firmware-brcm80211 firmware-realtek
5 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 7,283 kB of archives.
After this operation, 15.6 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://archive.ev3dev.org/debian stretch/main armhf bluez-firmware all 1.2-3+rpt4.1 [122 kB]
Get:2 http://archive.ev3dev.org/debian stretch/main armhf brickd armhf 1.1.2 [14.8 kB]
Get:3 http://archive.ev3dev.org/debian stretch/main armhf firmware-atheros all 1:20161130-3+rpt3 [3,132 kB]
Get:4 http://archive.ev3dev.org/debian stretch/main armhf firmware-brcm80211 all 1:20161130-3+rpt3 [3,673 kB]
Get:5 http://archive.ev3dev.org/debian stretch/main armhf firmware-realtek all 1:20161130-3+rpt3 [342 kB]
Fetched 7,283 kB in 0s (7,786 kB/s)
(Reading database ... 36125 files and directories currently installed.)
Preparing to unpack .../bluez-firmware_1.2-3+rpt4.1_all.deb ...
Unpacking bluez-firmware (1.2-3+rpt4.1) over (1.2-3+rpi2) ...
Preparing to unpack .../brickd_1.1.2_armhf.deb ...
Unpacking brickd (1.1.2) over (1.1.1) ...
Preparing to unpack .../firmware-atheros_1%3a20161130-3+rpt3_all.deb ...
Unpacking firmware-atheros (1:20161130-3+rpt3) over (1:0.43+rpi6) ...
Preparing to unpack .../firmware-brcm80211_1%3a20161130-3+rpt3_all.deb ...
Unpacking firmware-brcm80211 (1:20161130-3+rpt3) over (1:0.43+rpi6) ...
Preparing to unpack .../firmware-realtek_1%3a20161130-3+rpt3_all.deb ...
Unpacking firmware-realtek (1:20161130-3+rpt3) over (1:0.43+rpi6) ...
Setting up firmware-realtek (1:20161130-3+rpt3) ...
update-initramfs: deferring update (trigger activated)
Setting up bluez-firmware (1.2-3+rpt4.1) ...
Setting up brickd (1.1.2) ...
Setting up firmware-atheros (1:20161130-3+rpt3) ...
Setting up firmware-brcm80211 (1:20161130-3+rpt3) ...
```

dlech, 2018-04-15:
What is the output of `cat /proc/device-tree/model; echo`?

zombiezen, 2018-04-16:
```
Raspberry Pi 3 Model B Plus Rev 1.3
```

dlech, 2018-04-16:
Thanks. should be fixed now. Please try `sudo apt update && sudo apt upgrade` again.

dlech, 2018-04-20:
cool. please open a new issue if you run into any other problems.

closed 2018-04-20 by dlech