ev3dev/ev3dev #1058: ERROR:  ImportError: No module named ev3dev.ev3

clplaneguy, 2018-03-28:
Yesterday I ran a sample program to beep (Beep.py).  Today the program does not run.  I get the message "ImportError: No module named ev3dev.ev3"  I finally got PyCharm to connect to OpenCV cv2 (opencv-python).  But, now on further testing, I find out that the beep program no longer runs.  Now, how do I fix the beep program?

Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2018-03-28:
ev3dev-lang-python is python3 only and opencv is python2 only (at least in the current versions of ev3dev). See #803

clplaneguy, 2018-03-28:
The EV3 has both Python2 and Python3.

robot@ev3dev:~$ python2

Python 2.7.9 (default, Aug 13 2016, 17:33:18)

robot@ev3dev:~$ python3

Python 3.4.2 (default, Oct  8 2014, 14:47:30)

The real default is Python 2.7.9

robot@ev3dev:~$ python

Python 2.7.9 (default, Aug 13 2016, 17:33:18)

My simple beep program states in the first line that it is written in a
Python 3 environment.

*#!/usr/bin/env python3 **from *ev3dev.ev3 *import **
*# needed for Sound *Sound.beep()

So, to casually try to execute

>>>Python Beep.py

Is to be ignorant of the details.  This would to be trying to execute a
Python 3 script in Python 2.  Not a good idea.  It does not work.  To
properly execute this Python 3 Beep.py script in Python 3 one needs to …

>>>Python3 Beep.py

On another matter, the only reference OpenCV has to Python is to Python 2.7!
So, an OpenCV script may need to include

*#!/usr/bin/env python2    #?*

Or, just to be perfectly clear, if ev3dev-lang-python is exclusicely in
Python 3 and OpenCV’s only port to Python is in Python 2.7,  does that mean
that there is no way to use OpenCV on the EV3?

clplaneguy, 2018-03-28:
The EV3 has both Python2 and Python3.

```
robot@ev3dev:~$ python2
Python 2.7.9 (default, Aug 13 2016, 17:33:18)

```

```
robot@ev3dev:~$ python3
Python 3.4.2 (default, Oct  8 2014, 14:47:30)
```

The real default is Python 2.7.9

```
robot@ev3dev:~$ python
Python 2.7.9 (default, Aug 13 2016, 17:33:18)
```

My simple beep program states in the first line that it is written in a Python 3 environment.

```
#!/usr/bin/env python3
from ev3dev.ev3 import *  # needed for Sound
Sound.beep()
```

So, to casually try to execute

`>>>Python Beep.py`

Is to be ignorant of the details.  This would to be trying to execute a Python 3 script in Python 2.  Not a good idea.  It does not work.  To properly execute this Python 3 Beep.py script in Python 3 one needs to …

`>>>Python3 Beep.py
`

On another matter, the only reference OpenCV has to Python is to Python 2.7!  So, an OpenCV script may need to include

`#!/usr/bin/env python2    #?
`
Or, just to be perfectly clear, if ev3dev-lang-python is exclusicely in Python 3 and OpenCV’s only port to Python is in Python 2.7,  does that mean that there is no way to use OpenCV on the EV3?

dwalton76, 2018-03-28:
`does that mean that there is no way to use OpenCV on the EV3?`

The way I work-around this is by having a standalone python2 program that uses OpenCV.  I then do one of the following:
- call that program from my python3/ev3dev-lang-python program

closed 2018-07-15 by dlech