ev3dev/ev3dev #1059: brickpi3 and set_device for motors

dwalton76, 2018-03-28:
### What are you trying to do?
I am trying to use EV3 large motors and medium motors on a brickpi3 but cannot set the driver via `set_device`

### What actually happened?

```
robot@ev3dev:~/lego-crane-cuber$ echo lego-ev3-l-motor > /sys/class/lego-port/port4/set_device 
-bash: echo: write error: Operation not supported
robot@ev3dev:~/lego-crane-cuber$ ls -l /sys/class/lego-port/port4/
total 0
-r--r--r-- 1 root ev3dev 4096 Mar 28 15:57 address
lrwxrwxrwx 1 root root      0 Mar 28 16:40 device -> ../../../spi0.1
-r--r--r-- 1 root ev3dev 4096 Mar 28 15:57 driver_name
-rw-rw-r-- 1 root ev3dev 4096 Mar 28 15:57 mode
-r--r--r-- 1 root ev3dev 4096 Mar 28 15:57 modes
drwxr-xr-x 2 root root      0 Mar 28 16:40 power
--w--w---- 1 root ev3dev 4096 Mar 28 17:15 set_device
drwxr-xr-x 4 root root      0 Mar 28 16:40 spi0.1:MA:lego-nxt-motor
-r--r--r-- 1 root ev3dev 4096 Mar 28 15:57 status
lrwxrwxrwx 1 root root      0 Mar 28 16:40 subsystem -> ../../../../../../../../../class/lego-port
-rw-rw-r-- 1 root ev3dev 4096 Mar 28 15:57 uevent
robot@ev3dev:~/lego-crane-cuber$ 
```

The LargeMotors will still show as connected (apparently the `lego-ev3-l-motor` and `lego-nxt-motor`  drivers are either aliases for one another or considered close enough) but the MediumMotors will not show as connected because the driver is not `lego-ev3-m-motor`.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.9.59-ev3dev-1.6.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dlech, 2018-03-28:
The firmware in BrickPi3 (and earlier BrickPi models) doesn't know about different kinds of motors, so we just treat everything like an NXT motor. Does it actually make a difference to your program? I guess we are limiting the max speed too much when you do speed control?

dwalton76, 2018-03-28:
`Does it actually make a difference to your program` I don't think it will for this one (I don't think I use max speed anywhere), I will try making it a LargeMotor object instead.