ev3dev/ev3dev #1061: ev3dev-buildscripts apt-add-repository fails on stretch

dwalton76, 2018-03-29:
### What are you trying to do?
I am trying to follow the README at https://github.com/ev3dev/ev3dev-buildscripts but it barfs on

```
robot@ev3dev:~/ev3dev-buildscripts$ sudo apt-add-repository ppa:ev3dev/tools
Traceback (most recent call last):
  File "/usr/bin/apt-add-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)    
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for ev3dev/stretch
robot@ev3dev:~/ev3dev-buildscripts$ 
```

### What did you expect to happen?
no barfing

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.9.59-ev3dev-1.6.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dwalton76, 2018-03-29:
I'm able to get this going on my laptop so feel free to just close this if there is no long term goal of building the ev3dev kernel from an ev3dev machine.

dlech, 2018-03-29:
Only x86-64 Ubuntu LTS is supported.

closed 2018-03-30 by dwalton76