ev3dev/ev3dev #1064: IP Address not configuring properly with DHCP

computingnerd, 2018-04-05:
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

tcwan, 2018-04-06:
You need to have a DHCP server running on the PC. The 169.254.x.x is a auto-generated IP address when no DHCP server is available. See [this](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/)

computingnerd, 2018-04-06:
`192.168.137.x` was from my router's DHCP service (I don't have a DHCP on my laptop). For some reason it changed to `.137` on that day for some reason. Now its always `.1` so I can't use the Windows defaults. Is there anything wrong with my router's DHCP or do i need to download a DHCP service on my laptop?

dlech, 2018-04-08:
When you are using USB, the automatic IP address comes from Windows Internet Connection Sharing. The DHCP server built into Windows ICS breaks very easily. Generally, turning off Internet Connection Sharing and then turning it back on is enough to fix it.

If you router is also configured for the 192.168.137.x subnet and your computer is connected to your router, then Window will probably get confused because Internet Connection Sharing also uses the 192.168.137.x subnet. I would suggest using a different subnet on your router.

tcwan, 2018-04-08:
@computingnerd:
I don't use windows so I don't know the exact cause, but your EV3 getting the .137 subnet prefix could be due to your PC somehow changing the USB connection to Bridged mode rather than the default NAT configuration. Either way could work but configuring everything properly for Bridged mode is not trivial unless you're familiar with network troubleshooting.