ev3dev/ev3dev #1067: Large Motor reverses after rotation

worgarside, 2018-04-06:
I'm trying to get a motor to rotate accurately to 45 degree increments, up to 48 times in a row. After each rotation, the motor reverses by a couple of degrees and by the end of the loop, it's clearly out of line. I've tried several routes, such as adding a `ramp_down_sp` to slow down the end, or using `run_to_abs_pos` instead of relative, but to no avail. If I rotate the entire amount at once, the motor is perfectly aligned by the end, so it's a problem with a small overhead per rotation.

I'm using Python 3.6, with the latest version of ev3dev. This is for my dissertation so any help would be greatly appreciated!

I've uploaded a [video to demonstrate the issue here](https://youtu.be/IEwG2lL-Dww), and the code used in it is below. Thanks in advance!

```
from time import sleep
from ev3dev.ev3 import Sound, OUTPUT_A
from ev3dev.helper import LargeMotor

motor = LargeMotor(OUTPUT_A)
motor.stop_action = 'hold'

for i in range(16):
    motor.run_to_rel_pos(position_sp=200, speed_sp=510)
    motor.wait_until_not_moving()
    sleep(0.05)
    Sound.beep()
```

worgarside, 2018-04-09:
Apologies, this is the info:

```
Image file:         ev3dev-jessie-ev3-generic-2017-06-09
Kernel version:     4.4.68-20-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-04-09:
`run_to_rel_pos` is the command that should do what you are expecting. It is supposed to ignore any error in the position when called multiple times in a row.

I'm guessing that you are using `position_sp=200` to get 45° because of a gear ratio?

So the expected final position should be `200 * 16 = 3200`. What is the actual final position according to `motor.position`?

worgarside, 2018-04-09:
Yes the gear ratio is 1:40/9. I added a print statement to the loop for the value of `i` and `motor.position` and it returned the following:

```
0: 189
1: 384
2: 579
3: 778
4: 976
5: 1172
6: 1369
7: 1563
8: 1759
9: 1954
10: 2152
11: 2349
12: 2546
13: 2744
14: 2945
15: 3148
```

I tried removing the sleep statement but that had no discernible effect.

dlech, 2018-04-14:
Does this still happen using ev3dev-stretch?

dlech, 2018-04-14:
As a workaround, I get the expected results with this code:

```python
from time import sleep
from ev3dev.ev3 import Sound, OUTPUT_A
from ev3dev.helper import LargeMotor

motor = LargeMotor(OUTPUT_A)
motor.reset()
motor.stop_action = 'hold'
motor.speed_sp = 510

pos = motor.position

for i in range(16):
    pos += 200
    motor.run_to_abs_pos(position_sp=pos)
    motor.wait_until_not_moving()
    sleep(0.05)
    Sound.beep()
    print(motor.position)
```

worgarside, 2018-04-14:
I haven't had chance to test with stretch yet (and probably won't for a few weeks), but the workaround you suggested works perfectly. Thanks for you help! I'll get back to you when I test it on ev3dev-stretch