ev3dev/ev3dev #1068: How to detect an undocumented sensor?

jlehuen, 2018-04-08:
Hello,
I wrote a driver for the - low documented - NumericPad from Mindsensor (solved issue #1049) but I have a new question : _How can I detect if a NumericPad is connected to a specific port?_

Thanks,
Jerome

dlech, 2018-04-08:
Since this sensor doesn't follow the usual convention of having a vendor and product ID at registers 0x00 and 0x08, it will be hard to make a 100% positive identification. However, the default address of 0x5a is not used by any other MINDSTORMS compatible sensors that I know of. You could just try reading a byte from register 0x00 at that address. If there is no error, then it is pretty safe to assume that there is a keypad connected.

P.S. I'm guessing that when you say "I wrote a driver", that you don't mean a Linux kernel driver.

jlehuen, 2018-04-08:
Thank you for the advice. Sure, the code I posted at the end of issue #1049 is not a Linux kernel driver, just a function to read the sensor's data, sorry for the inaccuracy.
Jerome

jlehuen, 2018-04-08:
This code works does the job, thank you :

~~~c
#include <linux/i2c-dev.h>
#include "smbus.h"

bool keypad_is_plugged(int port) {
	char filename[20];
	snprintf(filename, 19, "/dev/i2c-%d", port+2);
	int file = open(filename, O_RDWR);
	if (file < 0) return false;
	if (ioctl(file, I2C_SLAVE, I2C_ADDR) < 0) return false;
	if (i2c_smbus_read_byte_data(file, DATA_REG) < 0) return false;
	return true;
}
~~~

closed 2018-04-08 by jlehuen