ev3dev/ev3dev #1069: Different program run speed on first run after boot

laurensvalk, 2018-04-13:
#### What are you trying to do?
Boot. Run a micropython or python3 program. Stop it, run it again. (Either through brickman, vscode, or ssh).

Specifically, [this program](https://github.com/laurensvalk/segway/blob/2bfa743840dc65fc5625b00441430c157f4359a8/ev3/ev3dev/pythonic/segway.py) on [this robot](http://robotsquare.com/2014/06/23/tutorial-building-balanc3r/).

#### What did you expect to happen?
The same thing, twice. In this case to make a robot balance itself.

#### What actually happened?
The first time the program runs more slowly. The second time it runs faster. The third time and onward are all the same as the second run. 

The first one is the odd one out. (Robot always collapses; runs fine the second or third time.)

#### What hardware and software are you using (including version numbers)?
```
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-ev3-generic-2017-12-13
Kernel version:     4.9.58-ev3dev-1.6.0-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533DDB29
BOARD_INFO_TYPE=main
```

laurensvalk, 2018-04-13:
So, this is a bit of a weird issue. I started noticing this when the segway would always collapse on the first run after boot. Works fine afterwards.

It took a while to figure that out, but it is extremely consistent. Happens after every boot.

It's probably not noticable for most programs.

No need to prioritize this issue, but I'm opening it so I can log any findings that may be relevant. (Constant rebooting is a bit time consuming.)

laurensvalk, 2018-04-13:
[ ] Check for unexpected values when reading the LEGO gyro for the very first time. Might cause initial balance problems.
[ ] Check if the initial destabilization disappears over the course of the initial run.

dlech, 2018-04-13:
maybe a duplicate of #1014?

dlech, 2018-04-13:
Does it make a difference if you wait some time after boot before the first run? Or does the first run happen ASAP after boot?

laurensvalk, 2018-04-13:
I tried waiting up to a minute or so and then the problem would still occur
on the first  run.

When running it twice within the first minute the problem would still not
occur the second time.

I'll try waiting longer next time, and time it to be sure.

The problem is that it is somewhat vague. It's clear when you see the robot
try to balance, but I haven't found a suitable metric (e.g. loop time or
busy time) to quantify it.

laurensvalk, 2018-04-13:
Are there hardware read/write buffer statistics that I can look at?

I'm wondering if there might be delays in the sensor data or written duty
cycle values (which would not affect the python loop time but would impact
the robot. )

dlech, 2018-04-13:
Not that I know of. When you write the duty cycle the PWM hardware is updated immediately. If you are using a UART sensor, there could be some buffering going on. But it is going on all of the time and I would not expect it to be affected by running a program.

laurensvalk, 2018-04-14:
Thanks for the input.

After some more reboots, it appears to be related to whether or not the gyro is put into calibrate mode.

closed 2018-04-14 by laurensvalk

dlech, 2018-04-14:
How are you putting the gyro sensor "into calibrate mode"?

laurensvalk, 2018-04-16:
By writing 'GYRO-G&A' to the mode file.

Specifically [like this](https://github.com/laurensvalk/ev3devlight/blob/dd143a25990b41487b5ee05e9274775ff0cb1abc/ev3devlight/sensors.py#L105) where [this is the mode property](https://github.com/laurensvalk/ev3devlight/blob/dd143a25990b41487b5ee05e9274775ff0cb1abc/ev3devlight/sensors.py#L32).

From some basic testing, this seems to reset both the accumulated angle and the rate bias to zero. 

But maybe this issue is related to the 'old' gyro type, which has problems switching between some modes.