ev3dev/ev3dev #1071: Does exist in Debian Stretch some utility to check batteryMonitor?

jabrena, 2018-04-16:
Hi @dlech,

Maybe it is a old question but I am not sure if anyone has developed something in last months.

Does exist some utility to check the battery and if the level reach some threshold in that case, shutdown the brick?

Many thanks in advance

Juan Antonio

jabrena, 2018-04-16:
I found a issue about it:
https://github.com/ev3dev/ev3dev/issues/861

What is the threshold level to shutdown?
https://github.com/ev3dev/brickd

How to configure?

Juan Antonio

dlech, 2018-04-16:
> What is the threshold level to shutdown?

See https://github.com/ev3dev/brickd/blob/ev3dev-stretch/src/power_monitor.vala#L115

> How to configure?

It is not configurable.

jabrena, 2018-04-16:
I am going to test this week.

Thanks @dlech

closed 2018-07-15 by dlech