ev3dev/ev3dev #1072: Only a moment can connect ev3 via bluetooth

aacco, 2018-04-19:
### What are you trying to do?
I try to connect ev3 to a PC via bluetooth.

### What did you expect to happen?
Sorry, I have no idea.

### What actually happened?
I type: connect <macaddress>
then the connection succeeded, but immediately below error displayed. 
### What hardware and software are you using (including version numbers)?
archlinux(Apr,19,2018now) 

below: log (<macaddress> is my ev3's macaddress)

[bluetooth]# connect <macaddress>
Attempting to connect to <macaddress>
[CHG] Device <macaddress> Connected: yes
[CHG] Device <macaddress> ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable
[CHG] Device <macaddress> ServicesResolved: no
[CHG] Device <macaddress> Connected: no

dlech, 2018-04-19:
This is normal. This `connect` command connects things like keyboards and audio and other simple services, but the EV3 does not provide any of these services. It connects just long enough to see that there is nothing to do and then disconnects.

The only thing ev3dev uses Bluetooth for is to create a network connection. 

http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

dlech, 2018-04-22:
I think :+1: means we can close the issue.

closed 2018-04-22 by dlech

aacco, 2018-04-22:
Thank you for your help!