ev3dev/ev3dev #1074: ev3dev-buster

dlech, 2018-04-22:
Collecting some notes here:

Debian buster will be released sometime in 2019: https://lists.debian.org/debian-devel-announce/2018/04/msg00006.html

The armel architecture (used on EV3) is likely to be dropped from official Debian support: https://release.debian.org/buster/arch_qualify.html

jabrena, 2018-04-22:
Yes, you have reason with the forecast, Debian releases every 2 years:
https://wiki.debian.org/DebianReleases?action=show&redirect=DebianRelease#Production_Releases

But Debian Stretch  is LTS and the support will finish in 2022 and Lego should release a new product in 2020.

**Lego Mindstorms Timeline:** 

1. January 1998: RCX
2. August 2006: NXT (After 8 years)
3. January 2013: EV3 (After 7 years)
4. January 2020: ?? (After 7 years)

Source: https://www.lego.com/en-us/mindstorms/history

Juan Antonio

dlech, 2019-03-04:
First ev3dev-buster snapshots are out: https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2019-03-04/

jabrena, 2019-03-04:
Any change in the EV3Dev interface?

dlech, 2019-03-04:
not yet...

jabrena, 2019-03-05:
Maybe we should not change the interface until Stretch is official. It requires many changes and it could be better to have some documentation for the migration or at least, the list of issues affecting the interface.

dlech, 2019-03-05:
There are no plans to make any changes to stretch.

dlech, 2019-04-15:
Latest news says armel (and armhf) will be in buster.

https://lists.debian.org/debian-devel-announce/2019/04/msg00003.html