ev3dev/ev3dev #1075: Potential motor controllers with ev3dev?

FaveAsian, 2018-04-22:
My robotics club this year got new ev3 bricks which I am using ev3dev with. Can anyone give me some recomendation on motor controllers that would work with the ev3 brick and tetrix DC motors. The language we are planning to is either python, java, or java with lejos if that helps.

dlech, 2018-04-22:
I don't know if anyone sells these anymore, but they will work with ev3dev.

http://www.hitechnic.com/blog/wp-content/uploads/HiTechnic-Motor-Controller-Specification.pdf

FaveAsian, 2018-04-22:
@dlech my club actually is using those now and we have several of those. The issue we have with those is that it isn't a simple plug and work. We have to insert the stripped cable which is highly inconvenient and were hoping for something simpler in terms of setup. Although I am glad to know that those will work for backup.

dlech, 2018-04-22:
If there are any other motor controllers compatible with your motors that use I2C or UART communications, then they could be made to work with ev3dev as well. You would have to make your own adapter cable to fit the EV3 though.

FaveAsian, 2018-04-22:
I am not to knowledgeable in the hardware category. Would you say these would work? http://modernroboticsinc.com/core-motor-controller-3 I would assume the dc motor is very similiar. I just need verification that this would work with ev3dev.

dlech, 2018-04-22:
Since that motor controller uses USB, you could plug it into the USB port on the EV3. However, it looks like they provide software for the Android OS, which is not compatible with ev3dev, so you would have to do more work to figure out how to program it.

FaveAsian, 2018-04-23:
Okay, I just have one last question. My club this year got the [DC Motor Expansion Controller](http://modernroboticsinc.com/core-motor-controller-3). They were trying to get lejos working on there, but ran into issues with lejos itself. I am wondering does it support lejos as from my own research, it seems like it supports arduino which I would then have to buy the[ main motor controller board ](https://www.tetrixrobotics.com/PRIZM). The other languages it supports was the lego mindstorm drag and drop which has no controller support. Overall the whole issue my club is having is due to the fact that the DC Motor Controller doesn't work with languages like robotC which my club has expirience in.

dlech, 2018-04-23:
There is no nice library on ev3dev (or LeJOS) that will let you say `motor.run(...)` using these controllers. You will have to find documentation on how these controllers communicate using a serial port. For example, for the HiTechnic motor controller, in the document I linked above, there is a table on page 5. It tells you which I2C memory registers to write to to perform certain commands or read certain data. This is what I mean by "low-level", knowing exactly what information is traveling over the wires to tell the controller what to do.

closed 2018-04-23 by FaveAsian