ev3dev/ev3dev #1077: How to enable Bluetooth on RPi 3B+ with BrickPi3?

zombiezen, 2018-04-26:
This might be a simple documentation issue, since I see that later in config.txt that it mentions that the device overlay tree lines for the BrickPi need to be uncommented.  Does the BrickPi3 also require me to scale the CPU frequency down to 250?

### What are you trying to do?

```shell
$ sudo systemctl start bluetooth.service
$ bluetoothctl
```

### What did you expect to happen?

Interactive shell

### What actually happened?

```
$ systemctl status bluetooth.service
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:bluetoothd(8)
$ bluetoothctl
Waiting to connect to bluetoothd...
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-04-14
Kernel version:     4.14.33-ev3dev-1.6.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=00000000b17b91a3
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=0996722E504D5741372E3120FF0E2A11
BOARD_INFO_TYPE=aux
```

zombiezen, 2018-04-26:
I uncommented the two lines in config.txt that seemed appropriate and rebooted:

```
dtoverlay=pi3-miniuart-bt
core_freq=250
```

This did not appear to change anything.

dlech, 2018-04-26:
The 4.14 kernel changed how bluetooth is configured for RPi's. I haven't really looked into it yet. I was hoping that it would just work, but I guess not.

You definitely don't want to do this:

```
dtoverlay=pi3-miniuart-bt
core_freq=250
```

It will cause the bluetooth and your RPi to operate at a reduced capacity. It is only needed with the original BrickPi and BrickPi+ because those devices use the same UART that is used by the built-in Bluetooth. BrickPi3 doesn't use the UART pins, so there is no problem there.

dlech, 2018-04-26:
also you should not need to manually start the bluetooth service

zombiezen, 2018-04-26:
Yup, I've reverted back to the old commented-out configuration. Just wanted to enumerate things that I've tried.

dlech, 2018-04-26:
Well, it looks like things actually didn't change, so I inadvertently broke it. Will be fixed in next kernel release.

https://github.com/ev3dev/rpi-kernel/commit/6bbf7f6f5866f6b5382add2cda371db7a92a6ae3

dlech, 2018-04-26:
Although I have only tested on RPi3, not RPi3+. So, if you could run

    sudo /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

and see if it gets bluetooth working, that would be helpful.

Tip: you may have to power on bluetooth using `connmanctl` before using `bluetoothctl`

    connmanctl enable bluetooth

dlech, 2018-04-26:
Looks like this needs more work.

Official RPi script:

```sh
#!/bin/sh

HCIATTACH=/usr/bin/hciattach
SERIAL=`grep Serial /proc/cpuinfo | cut -c19-`
B1=`echo $SERIAL | cut -c3-4`
B2=`echo $SERIAL | cut -c5-6`
B3=`echo $SERIAL | cut -c7-8`
BDADDR=`printf b8:27:eb:%02x:%02x:%02x $((0x$B1 ^ 0xaa)) $((0x$B2 ^ 0xaa)) $((0x$B3 ^ 0xaa))`

if [ "$(cat /proc/device-tree/aliases/uart0)" = "$(cat /proc/device-tree/aliases/serial1)" ] ; then
	if [ "$(wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins | cut -f 1 -d ' ')" = "16" ] ; then
		$HCIATTACH /dev/serial1 bcm43xx 3000000 flow - $BDADDR
	else
		$HCIATTACH /dev/serial1 bcm43xx 921600 noflow - $BDADDR
	fi
else
	$HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
fi
```

dlech, 2018-04-26:
https://github.com/ev3dev/rpi-kernel/commit/e5625d9814c19e2bd3c33a718c56e7b7c7b66d54

zombiezen, 2018-05-02:
Sorry, do you still need me to test that? Or have you already found the fix?

zombiezen, 2018-05-03:
I tried the following:

```
$ sudo /usr/bin/hciattach /dev/tty/AMA0 bcm43xx 921600 noflow -
Can't open serial port: Not a directory
Can't initialize device: Not a directory

$ SERIAL=`grep Serial /proc/cpuinfo | cut -c19-`
$ B1=`echo $SERIAL | cut -c3-4`
$ B2=`echo $SERIAL | cut -c5-6`
$ B3=`echo $SERIAL | cut -c7-8`
$ BDADDR=`printf b8:27:eb:%02x:%02x:%02x $((0x$B1 ^ 0xaa)) $((0x$B2 ^ 0xaa)) $((0x$B3 ^ 0xaa))`
$ cat /proc/device-tree/aliases/uart0
/soc/serial@7e201000
# output did not have ending LF
$ cat /proc/device-tree/aliases/serial1
/soc/serial@7e201000
# output did not have ending LF
$ wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins | cut -f 1 -d ' '
16
$ sudo /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow - $BDADDR
Can't open serial port: Not a directory
Can't initialize device: Not a directory
```

I suspect the answer may lie in the systemd unit alongside the official script. Notice how in [pi-bluetooth.hciuart.service](https://github.com/RPi-Distro/pi-bluetooth/blob/2a7477966bb3c69838b224f3ea92cb49a88124d5/debian/pi-bluetooth.hciuart.service) that it depends on /dev/serial1? I tried running `sudo systemctl service start dev-serial1.device`, but that seemed to hang.

dlech, 2018-05-03:
> sudo /usr/bin/hciattach /dev/tty/AMA0 bcm43xx 921600 noflow -

This should be:

    sudo /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

(no slash between `tty` and `AMA0`)

BertLindeman, 2018-05-04:
Looks like I am missing something here, on
` bluetoothctl` 
I get:
```bluetoothctl: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory```

Same procedure as @zombiezen above this.

Running:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-04-22
Kernel version:     4.14.34-ev3dev-2.0.0-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=0000000020e19c66
BOARD_INFO_TYPE=main
```

dlech, 2018-05-04:
@BertLindeman, I would suspect a corrupt filesystem/bad SD card.

BertLindeman, 2018-05-04:
New SD card build yesterday, but you never know, will try another.
Thanks.

BertLindeman, 2018-05-04:
Fresh copy of the image re-flashed.
You pointed me in the right direction, @dlech ;-) Of course.
Assume personal error somewhere.
Tx

dlech, 2018-05-05:
Fix is released in kernel `4.14.39-ev3dev-2.0.1-rpi2`.

closed 2018-05-05 by dlech