ev3dev/ev3dev #1081: Sound.speak("text").wait() terminating before finished

retrodaredevil, 2018-04-29:
### What are you trying to do?
Use the Sound.speak("text").wait() methods to do text to speech, then to pause the program until it is done saying "text"

### What did you expect to happen?
I expected it to wait until finished speaking.

### What actually happened?
Using wait() seems to resume the program before it is actually finished speaking.
Running these lines in a python3 shell caused this output:
```
>>> Sound.speak("hello there how are you").wait(); Sound.speak("moo").wait()
0
aplay: main:722: audio open error: Device or resource busy
0
>>> Sound.speak("hello there how are you").wait(); Sound.speak("moo hoo hi").wait()
0
aplay: main:722: audio open error: Device or resource busy
```
The first "hello there how are you" worked fine (note the 0 exit code) however the second one caused the error. Note the first time (with Sound.speak("moo").wait()) terminating immediately after it printed the error. However the second one with longer text "moo hoo hi" caused the program to stay like that forever until a KeyboardInterrupt.

Also note I tried the Sound.tone method and that seemed to wait until it was finished.

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-04-29:
cc @ev3dev/python

ddemidov, 2018-04-30:
https://github.com/ev3dev/ev3dev-lang-python/issues/461

dlech, 2018-04-30:
@retrodaredevil, so it looks like you can work around this by making your own `speak` function:

```python
    def speak(text, espeak_opts='-a 200 -s 130'):
        """
        Speak the given text aloud.
        """
        with open(os.devnull, 'w') as n:
            cmd_line = '/usr/bin/espeak --stdout {0} "{1}"'.format(espeak_opts, text)
            espeak = Popen(shlex.split(cmd_line), stdout=PIPE)
            play = Popen(['/usr/bin/aplay', '-q'], stdin=espeak.stdout, stdout=n)
        return play
```

closed 2018-05-25 by dlech