ev3dev/ev3dev #1087: server's down?

TechnoTew, 2018-05-03:
### What are you trying to do?
sudo apt-get update

### What did you expect to happen?
update?

### What actually happened?
Err http://archive.ev3dev.org jessie InRelease
  
Err http://security.debian.org jessie/updates InRelease                    
  
Err http://httpredir.debian.org jessie InRelease                           
  
Err http://archive.ev3dev.org jessie Release.gpg                          
  Temporary failure resolving 'archive.ev3dev.org'
Err http://security.debian.org jessie/updates Release.gpg
  Temporary failure resolving 'security.debian.org'
Err http://httpredir.debian.org jessie Release.gpg
  Temporary failure resolving 'httpredir.debian.org'

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-jessie-ev3-generic-2017-06-09
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-05-03:
You EV3 needs to be connected to the Internet. Which of these tutorials did you follow?

http://www.ev3dev.org/docs/networking/

dlech, 2018-05-03:
You can check your connection status by running `connmanctl status`

If it says `State = online`, you have an Internet connection. If it says anything else, then you don't.

TechnoTew, 2018-05-04:
@dlech There isn't a status command, but when i run `connmanctl state` i get 

State = ready
OfflineMode = False
SessionMode = False

TechnoTew, 2018-05-04:
I found out the reason, I forgot to on internet sharing 😞 disappointed in myself

sorry @dlech

closed 2018-05-04 by TechnoTew