ev3dev/ev3dev #1089: Streaming vids onto website to control robot

TechnoTew, 2018-05-05:
Something like seanbot, but for a beginner.
Any sites I can check out? It seems like the seanbot github has been taken down.

dwalton76, 2018-05-05:
I need to update the links in my seanbot blog posts but this is what you want 

https://github.com/ev3dev/ev3dev-lang-python-demo/blob/jessie/robots/EV3D4/EV3D4WebControl.py

dwalton76, 2018-05-05:
Some basic info on how to use it at 
http://programmablebrick.blogspot.com/2016/10/seanbot-20-smartphone-web-interface-for.html?m=1

dwalton76, 2018-05-05:
Let’s back up, are you running jessie or stretch?

TechnoTew, 2018-05-06:
jessie

TechnoTew, 2018-05-06:
dlech's says my webcam only works on stretch, ill migrate over when I have the time.

TechnoTew, 2018-05-06:
Got on stretch

I got my webcam to work, and stream on a port on motion
Webcam: Logitech QuickCam IM V-UAM27
However, Resolution stated is 640x480, but I can only get it to take pictures and stream at a maximum of 320x240, why is this so?

dlech, 2018-05-06:
> Resolution stated is 640x480, but I can only get it to take pictures and stream at a maximum of 320x240, why is this so?

As I mentioned in the other issue, this is probably because the EV3 USB port is USB 1.1 only.

Also, it looks like you might need to adjust the focus on the webcam.

TechnoTew, 2018-05-07:
What do u mean by adjusting the focus?

dlech, 2018-05-07:
On my webcams, I can rotate the part of the webcam that holds the lens clockwise or counterclockwise to focus it.