ev3dev/ev3dev #1090: Fswebcam newbie issues

TechnoTew, 2018-05-05:
### What are you trying to do?
fswebcam test.jpg

### What did you expect to happen?
no error and picture taken

### What actually happened?
--- Opening /dev/video0...
stat: No such file or directory

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

TechnoTew, 2018-05-05:
The webcam can be found through `lsusb`
Output: 
Bus 001 Device 002: ID 046d:08a0 Logitech, Inc. QuickCam IM
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

dwalton76, 2018-05-05:
Try ‘sudo fswebcam’

dlech, 2018-05-05:
Duplicate of #1030.

closed 2018-05-05 by dlech