ev3dev/ev3dev #1095: Error accessing sensors or importing?

TechnoTew, 2018-05-15:
### What are you trying to do?
Accessing infrared sensor

### What did you expect to happen?
No error occuring

### What actually happened?
```
from: can't read /var/mail/time
from: can't read /var/mail/ev3dev.ev3
-bash: test.py: line 4: syntax error near unexpected token `('
-bash: test.py: line 4: `ir = InfraredSensor()'
```

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-stretch-ev3-generic-2018-04-22
Kernel version:     4.14.38-ev3dev-2.0.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533DD4F5
BOARD_INFO_TYPE=main
```

dlech, 2018-05-15:
Tip: when you have a problem with a program, always post the source code of the program.

Your python file needs a "shebang" at the beginning of the file:

    #!/usr/bin/env python3

Otherwise it will try to run as a bash script, which is how you are getting those errors.

TechnoTew, 2018-05-17:
```
#!/usr/bin/env python3

from time import sleep
from ev3dev.ev3 import *
ir = InfraredSensor()
while True:
    print(ir.value)
    sleep(1)
```
Code is here but already has a shebang

TechnoTew, 2018-05-17:
Got it to work finally, not because i didnt have the shebang, but because i didn't manage to connect and assert the sensor.

closed 2018-05-17 by TechnoTew