ev3dev/ev3dev #1096: Open Roberta Lab Hangs

TechnoTew, 2018-05-19:
### What are you trying to do?
Open Roberta Lab display values

### What did you expect to happen?
Able to quit program normally

### What actually happened?
Ev3 is stuck in the program and can't quit, if program ends, blank screen

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-stretch-ev3-generic-2018-04-22
Kernel version:     4.14.38-ev3dev-2.0.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533DD4F5
BOARD_INFO_TYPE=main
```

dlech, 2018-05-19:
cc: @ensonic

dlech, 2018-05-25:
I would suggest opening an issue at https://github.com/OpenRoberta/robertalab-ev3dev

TechnoTew, 2018-05-28:
Ok @dlech

dlech, 2018-05-28:
https://github.com/OpenRoberta/robertalab-ev3dev/issues/64

closed 2018-07-15 by dlech