ev3dev/ev3dev #1099: Request to package OpenJDK10 on Debian Stretch

jabrena, 2018-05-23:
Hi @dlech,

Traditionally, `Java` users for `EV3Dev` installed `Oracle JRE 8` from the website:
http://www.oracle.com/technetwork/java/embedded/downloads/javase/javaseemeddedev3-1982511.html

But with the help of @JakubVanek, we could build `OpenJDK 10` for `EV3` and now the compilation run in the official **AdoptOpenJDK** build farm:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk-10-ev3-pipeline/

I think that it could be useful if we could package the JVM for the Java EV3Dev users. 

Cheers

Juan Antonio

dlech, 2018-05-23:
Openjdk-9 is available for stretch in debian backports already. What is the advantage of openjdk-10?

jabrena, 2018-05-23:
Hi mate,

Every release, the JVM internally offers new performance improvements but besides offers some improvements in terms of syntax. I put this link with more details:
http://www.oracle.com/technetwork/java/javase/10-relnote-issues-4108729.html

The real improvement with this artifact is the usage of `jlink` to create small Java Runtime images with the minimum Java packages and runtime utilities to offer a minimalistic runtime experience to run any Java program for EV3Dev. 

This is the reason that the size is only 21.67MB against the classic 100MB of `ejdk-8-fcs-b132-linux-arm-sflt-03_mar_2014.tar.gz`

jabrena, 2018-05-23:
If you like, we could give you a .sh for the packaging process.

jabrena, 2018-05-25:
Hi @dlech, 

I am certificating the `JKD` in the same way that any JDK included in any production server in the world. 

When this issue is resolved:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/466 
The build for `EV3Dev`, will pass all quality standards. 
I am working on it. 

Cheers

Juan Antonio

JakubVanek, 2018-08-18:
Hmm, I think we should create our own packaging, similarly to how I've done it in http://github.com/ev3dev-lang-java/openjdk-system. Then we should think about where/how to build the package and where to host it.

There are few assumptions the ev3dev-lang-java/openjdk-system project makes:

- CDS cache is ready and up-to-date,
- /usr/bin/java points to the packaged java, this is probably best achieved by using update-java-alternatives.

dlech, 2018-08-18:
You can read about packaging for ev3dev here: https://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/

There is no need to create a debian package repo, we already have one and I can give you upload access.

But it will be up to you guys to create and maintain the debian package. I'll be glad to answer questions though.

jabrena, 2018-08-18:
Many thanks @dlech,

We will review the documentation to create a Debian Project for EV3Dev Stretch.
https://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/

Cheers

Juan Antonio

jabrena, 2018-08-20:
Hi @dlech,

maybe the first step for the packaging process could be the creation of a new repository. In this case, the repository will store the software to package a JVM specifically designed and built for EV3. So:

What do you think about this name: @dlech & @JakubVanek 

- openjdk-ev3-deb

Can you say one example about another repository packaged in debian format to review the structure.
Apparently, if we store the build in the repo and add a bash script to install, maybe we could have the first version.

Juan Antonio

JakubVanek, 2018-08-20:
I have seen at least https://github.com/ev3dev/ev3-config

dlech, 2018-08-20:
My inclination would be to fork the [debian openjdk-10](https://tracker.debian.org/pkg/openjdk-10) package and work from there. We can remove all of the binary packages and add our own new one based on `-headless`.

Also, I would like to build this package for all ev3dev architectures, not just EV3.

jabrena, 2018-08-20:
Hi @dlech, 

in EV3Dev, the unique platform with bad JVM support is EV3 not Raspberry. In relation to EV3, we have the JVM ready to package in EV3Dev from the Continuous Integration infrastructure from AdoptOpenJDK:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_build_ev3_linux/

This is the use case that we want to cover

Juan Antonio

dlech, 2018-08-20:
So why can't we just apply the patches you have made to the existing debian package? This seems simpler than trying to make a new conflicting package for the same thing and trying to deal with alternatives.

jabrena, 2018-08-20:
mmmm, it is reasonable your comment. :)

The question is the size of the package.
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_build_ev3_linux/

A JDK requires 260MB:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_build_ev3_linux/lastSuccessfulBuild/artifact/build/jdk-ev3.tar.gz

And our minimal JRI (Java runtime image) only requires 21MB and of course and it is headless and includes some commands to add RMI support for the future.

I think that after some months building and using the JVM for EV3, we feel confident with the solution and the weight is minimum.

This was the first issue about OpenJDK build for EV3 + EV3Dev:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/232
https://github.com/ev3dev-lang-java/openjdk-ev3/releases/tag/v0.2

After Java 8, it is very complex to install Java on EV3. This is the reason that we created this initiative. We could start with current solution and with the user feedback scale but we have a solution running and tested with the libraries and itself with the same tools that other JVM:
https://github.com/AdoptOpenJDK/openjdk-tests
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_openjdktest_ev3_linux_full/
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_openjdktest_ev3_linux_full/64/testReport/

dlech, 2018-08-20:
So the runtime is usable without the 68MB jmods-ev3.tar.gz also? This would be a good reason to have the same package for the other architectures as well.

jabrena, 2018-08-20:
yes, in my case I am using the jri only at the moment.

I believe that we should start with EV3 only, we don´t have many hands to maintain many platforms.
With a Raspberry Pi 3 Model B+ it is possible install a JVM in seconds but the same experience in EV3 is a nightmare. With a good documentation is enough at the moment.

But I think that the best feedback is the user feedback.

We could start with EV3, offer a solution to improve the user experience and enrich the solution later with the testimonial from final EV3 user, in my humble opinion.

JakubVanek, 2018-08-20:
We have the ability to build Java for all ev3dev platforms (ev3, rpi1, rpi2, rpi3). It is only that we do not have Jenkins build jobs for them. Later we'll probably differentiate the platforms by a package name.

Quick intro to our build process:
1. ev3dev armel/armhf cross docker image is created and entered
2. [SoftFloat-3e fork](https://github.com/ev3dev-lang-java/softfloat-openjdk) is built for some floating point precision workaround from earlier times (http://mail.openjdk.java.net/pipermail/aarch32-port-dev/2016-December/000671.html). The fork could be replaced with original version if an openjdk adapter is moved from the fork to OpenJDK.
3. Patched OpenJDK is built.
4. The JMOD archive and the JRI are created from the built JDK. This step can be performed on any JDK.

The patches do this:
* they add `-march`/`-mtune` flags for each platform
* they add option for compiling with softfloat-3e
* they turn off multiprocessor support and some memory barriers for ARMv5/EV3

I see we have these options:
* Fork the existing OpenJDK debian package and implant our build process (without Docker) into it.
* Create a new package which would only carry binaries, i.e. it would take the JRI archive and package it.
* Create a new package which would also wrap our build process.

Regarding the java-alternatives: we do not need to execute `update-alternatives`. Installing `/usr/lib/jvm/.jri.platform.version.jinfo` (and then maybe calling `update-java-alternatives --auto`) would be sufficient.

jabrena, 2018-08-21:
Hi @JakubVanek ,

yes, but although we have the build process, later, it is necessary to test all builds and that process requires more time than the build process. This is the reason of previous comments. We have a solution for EV3, **if we dedicate to maintain JVM how develop robotics libraries?** we have few hands. The real problem for JVM is the EV3. For raspberry pi, it is better to have other alternatives and indeed if the installation of the JVM would take few time in EV3, it could be reasonable to install with apt, but this is not the case. for Raspberry PI, it is possible to install easily in few seconds because the hardware has better CPU and memory.

In terms of maintenance @JakubVanek, at the moment we have 4 jobs for EV3 (OpenJDK11), if we add more support, we have to maintain 12 additional jobs. **Do you have that time to maintain clean and updated?** This is the question mate. Sometimes, it is better to start with small things only.

Juan Antonio

JakubVanek, 2018-08-21:
Hmmm, to be honest:
* We do not need to build openjdk9 and openjdk10 for each platform. We need only the current stable release build and maybe a bleeding-edge openjdk build. _This is one (two) jobs per platform._ Moreover, I have scheduled the current jobs to run periodically once per week.
* We do not necessarily need to do such a testing that a single run takes a day and _runs in QEMU_ (i.e. openjdktests, systemtests; but you wanted it, so I helped). And if I understood correctly what we were doing, we were running the tests and blacklisting the failing ones until everything was green. And more often than not the results are not consistent even for the same tested build. So why the hell do we have it if we do not do anything with it? If we create one small specialized test job, _this is one job per platform_. If we integrate it with the first build using a top-level pipeline, this one can be automated too.
* I do not want to develop the library. Currently I want to finish a few things and then minimize the work to periodically releasing some package and fixing critical bugs.
  * openjdk-system integration,
  * java debian packaging,
  * library debian packaging,
  * template project integration with openjdk-system,
  * upstreaming of some of our patches.

JakubVanek, 2018-08-21:
Hmmm, if the debian package builds for Raspberry Pi use ARM32 JIT, we can skip building our own Java and fast forward to creating the JMOD archive and JRI.

JakubVanek, 2018-08-21:
Quick search about upstream status:
* EV3: nope, JIT is not a supported configuration on ARMv5, so probably only Zero armel build is applicable.
* RPi1: nope, JIT is not a supported configuration on ARMv6, so probably only Zero armel build is applicable.
* RPi2: yes, ARMv7 armhf is a supported configuration.
* RPi3: yes, ARMv7 armhf is a supported configuration.

jabrena, 2018-08-21:
So, we should start with EV3, receive feedback and evolve. But start with a small area and grow but little by little. :)

JakubVanek, 2018-08-21:
I have discussed this with Juan and we have reached a conclusion:
* We will start with EV3 package and finish it. It would be used for RPi1 as well, if there are not many users requesting fast floating point math.
* Later, we can create single JRI from Debian ARMv7 Java for RPi2, RPi3 and EVB to a separate package.

dlech, 2018-08-21:
OK, so basically, you want to take the `jri-ev3.tar.gz` tarball from Jenkins and turn it into a debian package. Do you also need to package `jmods-ev3.tar.gz`?

There are very few users of ev3dev on RPi 0/1, so I think we can just do EV3 for now as you have suggested.

JakubVanek, 2018-08-21:
No, jmods archive is only for users who would want to build their own JRI. They are contained in the JDK archive as well, but this way they user doesn't have to download the full JDK for EV3.

dlech, 2018-08-21:
OK. Are all of the binaries and configuration files in `jri-ev3.tar.gz` relocatable or will you need to rebuild with different options? /bin/ should be /usr/bin/, /lib/ should be /usr/lib/ and /conf/ should be /etc/java-10-openjdk-ev3dev/ (unless these files are not actually meant to be edited).

If we want this to co-exist with other java runtimes, then it gets more complicated, but I don't think that is really necessary at this point.

JakubVanek, 2018-08-21:
Hmm, I think we can follow the directory model of Ubuntu's openjdk packaging - they effectively "unpack" java into `/usr/lib/jvm/java-<ver>-openjdk-<arch>`. To achieve configurability, they simply symlink individual files from the `conf/` and `lib/` directories to the `/etc/java-11-openjdk` directory. Coexistence with other Javas then becomes easier. I have quickly looked at ubuntu's postinst/preinst hooks and there they install the alternatives using `update-alternatives`.

JakubVanek, 2018-11-11:
Hi,

I have added support for generating Debian packages as part of the build process - packaging files can be found [here](https://github.com/ev3dev-lang-java/openjdk-ev3/tree/master/packaging). If the Jenkins build succeeds, [openjdk11_build_ev3_linux](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_build_ev3_linux/) will contain the `debian.zip` artifact, which will contain the result of `debuild -us -uc`.
However, I'm not following the standard way of creating packages - the changelog is not preserved, it will always contain only the latest entry. Is it OK? If it is not good, I can maintain a separate git repository with the debian directory. Also, as we already have a separate build mechanism, the orig package is simply the `jri-ev3.tar.gz` archive.
This way, it should be possible to just sign the contents of `debian.zip` and upload them using dput.

@jabrena could you please review the package? It should be enough to download it to the brick and install it using `sudo apt install ./jri.deb`: [jri.deb.zip](https://github.com/ev3dev/ev3dev/files/2569825/jri.deb.zip) (GitHub doesn't let me upload .deb). You will then need to switch to the packaged java using `update-java-alternatives -s jri-11-ev3`.
@dlech could you please add me to the repository? My keys are the following: [SSH key](https://github.com/ev3dev/ev3dev/files/2569801/jakubvanek.ssh.txt), [GPG key](https://github.com/ev3dev/ev3dev/files/2569802/jakubvanek.gpg.txt).

Regards
Jakub Vaněk

jabrena, 2018-11-11:
hI @JakubVanek, 

tomorrow, I will test the package  both debian packages:

- Generated: https://github.com/ev3dev/ev3dev/files/2569825/jri.deb.zip
- Artifact from AdoptOpenJDK: https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_build_ev3_linux/

JakubVanek, 2018-11-11:
CDS dump will be executed automatically when the package is installed: [debian/postinst](https://github.com/ev3dev-lang-java/openjdk-ev3/blob/e46eefc3dfe3423d9768c5d37444c6e87dfd559a/packaging/debian/postinst.in#L73)

Yes, later we should decouple it from the main build. In fact, if the "fake" changelog is problematic, we will have to do it manually (now we have to sign it with gpg and publish it through dput anyway).

jabrena, 2018-11-12:
Hi @dlech,

I could install the debian package on my testing EV3 Brick few hours ago.
You have the details here:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/528

Can we continue with the process?

we would like to test a new EV3Dev Debian Stretch image with the Debian package installed.
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/

Cheers

Good job @JakubVanek 

Juan Antonio

dlech, 2018-11-13:
> the orig package is simply the `jri-ev3.tar.gz` archive.

FYI, not having a version number in this file name will be a problem. The first upload will work, but the next one will fail because of the duplicate file name. Although if the "source" package doesn't actually contain the source code, then there isn't really a reason to upload it. You can just upload the .deb files and skip the .dsc and .orig.tar.gz files.

> could you please add me to the repository?

done

JakubVanek, 2018-11-13:
Regarding the `jri-ev3.tar.gz` - I didn't tell the whole story - we are repackaging the contents of the archive to a `jri-11-ev3_11.0.1~13.orig.tar.xz` and alike, it's just that the software inside is binary only. Thanks for the advice though. I will not upload the source package, as it really would not make much sense.

Thank you for adding me to the repository.

Jakub

closed 2019-04-08 by dlech