ev3dev/ev3dev #1101: Difference in motor.position in run_forever not corresponding to run_to_rel_pos()?

obo, 2018-05-23:
### What are you trying to do?

I am trying to calibrate by tank robot to know how much rotation of left and right motors (in opposite directions) is needed to achieve the rotation of 180 degrees. I use IR beacon to know when the tank has fully rotated. Essentially I run:

```
left_motor.run_forever(speed_sp=speed)
right_motor.run_forever(speed_sp=(-1.0)*speed)
```
and then in a loop:
```
irangle = ir.value(0)
lpos = left_motor.position
rpos = right_motor.position
```
Whenever the irangle is at zero (and the *_motor.position() is far enough from the last reported value, I report the difference in lpos and rpos during this loop.

The measurements are pretty stable even across various speed_sp values ("At (speed value), " in this log):
```
At 100 , travel for 1/2circ  403 L and 400 R; absL: 805 , absR: 800
At 100 , travel for 1/2circ  401 L and 399 R; absL: 1206 , absR: 1199
At 100 , travel for 1/2circ  401 L and 400 R; absL: 1607 , absR: 1599
...
At 150 , travel for 1/2circ  405 L and 404 R; absL: 2821 , absR: 2805
At 150 , travel for 1/2circ  401 L and 398 R; absL: 3222 , absR: 3203
At 150 , travel for 1/2circ  401 L and 398 R; absL: 3623 , absR: 3601
...
At -180 , travel for 1/2circ  403 L and 404 R; absL: 2008 , absR: 1975
At -180 , travel for 1/2circ  405 L and 409 R; absL: 1603 , absR: 1566
At -180 , travel for 1/2circ  403 L and 405 R; absL: 1200 , absR: 1161
```

In short, it seems that going +402 with left and -402 with the right model should rotate the tank by 180 degrees.

When I tried using these values with run_to_rel_pos, the robot rotated not by 180 degrees but only by 20 degrees.

I now calibrated manually using a sequence of run_to_rel_pos calls and observing the values that accummulated in motor.position, it seems that 360 degrees is about 2100 steps, not 2*402.

### What did you expect to happen?

I expected that distances travelled with run_forever and run_to_rel_pos are identical.

My calibration loop relying on run_forever and getting position when IR indicates the angle of zero could be perhaps slow and thus off-sync with the real movement of the tank, but I kept the tank taking full two rounds and all four differences in motor.position were identical.

### What actually happened?

Distances travelled by run_forever seems *much* larger than distances travelled by run_to_rel_pos.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-04-22
Kernel version:     4.14.38-ev3dev-2.0.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535F74D2
BOARD_INFO_TYPE=main
```

obo, 2018-05-26:
The programs are working from the software point of view (and they are big and messy). This is more of a 'hardware' question if someone has seen a similar behaviour. I have solved it for now by manually running run_to_rel_pos with various position differences, until I found the value needed for a 360 turn of the tank.

Feel free to close this issue, I can reopen it if I manage to make a small working example program that demonstrates the difference in physical rotation of the tank for the same difference of values in motor.position when running with run_forever and the run_to_rel_pos input value.

ddemidov, 2018-05-27:
Just wondering, do you reset position to zero before run_forever?

obo, 2018-05-27:
I did not reset but I always polled the position and did the difference myself.

Attached is the code that waits for beacon and then starts spinning the robot at various speeds (and directions), always running left and right motors in opposite directions.

It spits the log lines as show above "At <speed>, travel for 1/2circ [takes] <left motor diff since last report> L and <right motor diff since last report> R; absL: <abs left position> , absR: <abs right position>

[find_angle_speed.zip](https://github.com/ev3dev/ev3dev/files/2042948/find_angle_speed.zip)

Also, the abs. values keep growing a lot if the random speed/direction choices are always for the same direction, but the reported diffs stay stable. So the problem is surely not due to measuring from a shifted zero.