ev3dev/ev3dev #1105: BrickPi3 + RaspberryPi model B+ Python programming

raphmc, 2018-05-28:
Hi there,

I read all the resources I could find but am still not really smarter. I posted in the ev3dev-lang-python but didn't get a answer so far so I figured you might be able to help me out!

I am currently at the beginnings of a very ambitious project and I struggle to find the right way to program my BrickPis.

I want to build a car manufacturing supply chain with an assembly unit. My model will have 2 suppliers (vertical warehouses), one assembly unit and a Lego PowerFunctions train to connect the different modules.
The orchestration of all the different production steps will be done in the cloud. This is why I chose to go for the BrickPi since the Raspberry gives me easy connectivity.

But I don't know how to best program my robots. The different options I found are:
– The Dexter Industries based Raspbian Jessie distribution. However, I find the programming rather difficult and unintuitive (from the examples I saw. I couldn’t test it yet since it does not run on the latest Raspberry Pi B+. But it looks like there is a Beta out that will offer support.)
– EV3DEV Jessie distribution (couldn’t test it for the same reasons). Is EV3DEV2 the new standard? Does it work well with the BrickPi3 and the newest RaspberryPi?

In your eyes, what is the easiest way to program my robots?

I hope you can help me make the right choices from the beginning and not realize halfway through the project that I went with the wrong setup :)

Thank you and have a nice weekend!

dlech, 2018-05-28:
> Is EV3DEV2 the new standard?

I assume you mean the ev3dev-lang-python module? I would call it new "experimental", not standard.

> Does it work well with the BrickPi3 and the newest RaspberryPi?

You need to use ev3dev-stretch if out are using RPi 3 B+.

> In your eyes, what is the easiest way to program my robots?

Using tools you are already familiar with. Or if you haven't done anything like this before, then using tools that your friends can help you with.

raphmc, 2018-05-29:
Thanks for your answer!
Unfortunately, I am not familiar with any of the tools and sadly I do not have any friends ;)
From what I understand, programming my robot in Python using the ev3dev-lang-python module appears to be far easier than using the Python library for the BrickPi on Raspbian. 
However, I know that I can to all the cloud integration on Raspbian and am not sure about ev3dev since I have not much experience with it.
So assuming I start at zero, no friends to help me, looking to control my robots via AWS cloud, should I go with Raspbian or ev3dev ?

dlech, 2018-05-29:
Raspbian and ev3dev are both based on Debian, so there really are not any significant differences other than how you program robots. They may have different software installed by default, but the same software, for the most part, is available for both. It sounds like ev3dev-lang-python sounds better to you, so why not go for it?

raphmc, 2018-06-01:
@dlech I managed to make ev3dev run with the BrickPi3 and have a robotic arm up and running. Looks to me like it is the right choice so far :)
Thanks for helping me out with it. 
I will see if I can write together a detailed explanation on how to make it work, since the documentation is not complete with regard to the BrickPi. I have not really experience with GitHub but if I find the time I will submit a pull request with an update.

dlech, 2018-06-01:
Cool. The easiest way to do documentation would be to just put something on the wiki.