ev3dev/ev3dev #1106: Speed up file access with f2fs

cho934, 2018-06-02:
### What are you trying to do?
Is it possible to speed up the file access with f2fs like that ?
http://whitehorseplanet.org/gate/topics/documentation/public/howto_ext4_to_f2fs_root_partition_raspi.html
Is it relevant ?
Is there another way to decrease the latency where I use ssh connection by wifi ?

### What did you expect to happen?
Speed up sd writing for example.

cho934, 2018-06-04:
Thank you for this explanation.
And what about the file Read/Write access ? using logs for example ? or file config access ?
What is your opinion ?

dlech, 2018-06-04:
Unless you are logging massive amounts of data, I don't think the file system makes much of a difference.

cho934, 2018-06-04:
For motioncontrol, writing text every 0.1sec in 2 files?

dlech, 2018-06-04:
My guess is that it would only make a small difference, but the only way to know for sure it to try it.

dlech, 2018-06-14:
FYI, I've enabled the f2fs module in kernel `4.14.49-ev3dev-2.0.2-ev3`.

cho934, 2018-06-14:
Some performance information  here...
https://lwn.net/Articles/520003/

cho934, 2019-04-09:
Is it always enabled with ev3dev stretch kernel ?

cho934, 2019-04-09:
All tasks are done. But I didn't succeed to boot because I missed the last step
http://whitehorseplanet.org/gate/topics/documentation/public/howto_ext4_to_f2fs_root_partition_raspi.html
In the EV3, how can I modify the rootfstype to change it from ext4 to f2fs ?
There is no cmdline.txt like in a RPI....
?

cho934, 2019-04-09:
I tried unsuccessfully to add int the uEnv.txt file the following bootargs 
```
bootargs="root=/dev/mmcblk0p2 rootfstype=f2fs"
```

dlech, 2019-04-09:
You are on the right track. Try `optargs=...` instead.

However, unlike raspberry pi, ev3dev has the `/boot` folder in the rootfs. So you will need to created a separate ext4 partition with `/boot` in it so that U-Boot can still read it.

You can also define `uenvcmd=...` here to completely override the boot script.

You can find the default boot script here:

https://github.com/ev3dev/u-boot/blob/03bd83c5c390feccfec335d47008267d30316804/include/configs/legoev3.h#L81-L129

dlech, 2019-04-09:
Basically, we want to end up with something like this:

```
   mmcblk0            60,1G              
   ├─mmcblk0p1 vfat     60M /boot/flash  
   └─mmcblk0p2 ext4     60M /boot   
   └─mmcblk0p3 f2fs     60G /   
```

cho934, 2019-04-09:
So I create them
```
sdg             7,4G                                                 
├─sdg1 vfat      48M /media/pmx/EV3DEV_BOOT                          EV3DEV_BOOT
├─sdg2 ext4      60M /media/pmx/ed34f903-0931-44e1-9064-f10e5fb98bdc 
└─sdg3          7,3G                                                 
```
But the 3rd partition in f2fs doesn't mount automatically since I create it and copy all files.
I filled the Uenv.txt with
```
optargs="root=/dev/mmcblk0p3 rootfstype=f2fs"
```

But because of the second partition, I think it's not enough? what else ?

cho934, 2019-04-09:
Just to be sure
mmcblk0p1 (vfat) contains boot.scr, u-boot.bin, uenv.txt
mmcblk0p2 (ext4 contains dtb..., vmlinuz..., initrd..., config..., system... files.
mmcblk0p3  (f2fs) contains all files remaining.

Am I right ?

dlech, 2019-04-09:
Yes.

Digging a bit more, this script is running: https://github.com/ev3dev/flash-kernel/blob/ev3dev-stretch/bootscript/all/bootscr.uboot-generic

So, `bootargs=...` is probably right.

Also with the new partitioning, you will need to add something like `prefix=/` to uEnv.txt.

cho934, 2019-04-09:
I don't find how to configure this prefix.
before it was /boot in the same /dev/mmcblk0p2 
now it will be /boot in the /dev/mmcblk0p2  while the / will be /dev/mmcblk0p3

dlech, 2019-04-09:
It will get mounted to `/boot` when Linux loads, but in the partition itself, all of the files should just be in the root `/` directory.

cho934, 2019-04-09:
```
bootargs="root=/dev/mmcblk0p2 rootfstype=f2fs prefix=/"
```
It seems to work because of the screen "EV3DEV" appeared then some message and finally an error:
```
mount : no such file or directory 
run-init : opening console : no such file or directory 
Target filesystem doesn't have requested /sbin/init
run-init : opening console : no such file or directory 
...
No init found try passing  init bootarg
```
The number of the partition is normally the same : 
bootpart=1
distro_bootpart=2

An idea ?

cho934, 2019-04-09:
I've just sending you the link to the video by email to better follow the error messages

dlech, 2019-04-09:
hmm... my guess would be that `/etc/fstab` needs to be updated. Also I think we are using an initrd, so that would need to be updated to included the changed fstab file. Or you would need to make a custom boot command that doesn't use the initrd.

cho934, 2019-04-09:
Ah yes of course... forget the 2nd partition

So my file /etc/fstab is the following:
```
#<file system>  <mount point>   <type> <options>                           <dump> <pass>
/dev/mmcblk0p1  /boot/flash     vfat   defaults,errors=remount-ro,noatime  0      3
/dev/mmcblk0p2  /boot     		ext4   defaults,errors=remount-ro,noatime  0      2
/dev/mmcblk0p3  /               f2fs   defaults,errors=remount-ro,noatime  0      1
proc            /proc           proc   defaults                            0      0
tmpfs           /run            tmpfs  nosuid,noexec,size=20M,nr_inodes=4096 0    0
```
But I received the same error when booting...

dlech, 2019-04-09:
probably because you didn't update the initrd or use a custom boot command to avoid the initrd.

cho934, 2019-04-09:
initrd is not important ?

dlech, 2019-04-09:
actually, it might be if the f2fs kernel driver is compiled as a module

cho934, 2019-04-09:
ok. I dont' understand what I have to do exactly with "update initrd". I tried to find unsuccessfully a boot parameter.

Let me know if you need to compile in the kernel mode. Then I will try again.

dlech, 2019-04-10:
Give this a try: https://oss.jfrog.org/artifactory/oss-snapshot-local/org/ev3dev/brickstrap/2019-04-08/f2fs.img.xz

Basically, what I did is updated `/etc/fstab` as above , added `f2fs` and `crc32` to `/etc/initramfs-tools/modules`, ran `sudo update-initframfs -u` and added `prefix=/` to `/boot/flash/uEnv.txt`.

FWIW, the boot time is about 10 seconds longer with f2fs, so I'm not really sold that making the whole rootfs use f2fs is a good idea. It could be better to just create a partition and mount it in `/home/robot/fast` or something like that.

dwalton76, 2019-04-10:
> Speed up sd writing for example

You could use a SSD but that will a decent amount of bulk.  I did this once with my brickpi using this board:
https://www.amazon.com/gp/product/B073PZ9X9W/ref=oh_aui_search_asin_title?ie=UTF8&psc=1

and this SSD:
https://www.amazon.com/gp/product/B00K67E5DA/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

It did boot pretty quickly :)

cho934, 2019-04-10:
> > Speed up sd writing for example
> 
> You could use a SSD but that will add a decent amount of bulk. I did this once with my brickpi using this board:
> https://www.amazon.com/gp/product/B073PZ9X9W/ref=oh_aui_search_asin_title?ie=UTF8&psc=1
> 
> and this SSD:
> https://www.amazon.com/gp/product/B00K67E5DA/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
> 
> It did boot pretty quickly :)

Good id, but we still use the brick EV3 for now. When we will have new sponsors for our team we will be able to change with a more powerfull card with all adapters ;)

cho934, 2019-04-10:
> Give this a try: https://oss.jfrog.org/artifactory/oss-snapshot-local/org/ev3dev/brickstrap/2019-04-08/f2fs.img.xz
>
> Basically, what I did is updated `/etc/fstab` as above , added `f2fs` and `crc32` [...]

So I confirmed that there are many FAILED during booting!
Wifi doesn't work well, it failed to connect.

Using RPI and this method , we gained 30 to 40% of performance using this f2fs file system, even using the interface (Rasbian for instance, or the one for 3D printer). So I really think it can be a good idea to use it, especially with the low speed of the sdcard bus.
Some figures here : https://lwn.net/Articles/520003/

And we need to prepare some benchmark tests to be sure that it can be useful.

> It could be better to just create a partition and mount it in `/home/robot/fast` or something like that.

ok. This is a good  and quick idea.

cho934, 2019-04-11:
I found how to fix the f2fs.img.xz issue.

In fact the fstab is not correct using f2fs
See here
https://www.funtoo.org/F2FS_Install_Guide

the file /etc/fstab need to be fixed with default f2fs options :
```
...
/dev/mmcblk0p3   /   f2fs    rw,acl,active_logs=6,background_gc=on,user_xattr           0 1
...
```

Then It works well, First time is a little bit longer (because of the fsck ?). Then second boot seems to be as long as usually (like the normal ext4 partition).

Second point. I had no problem to create a specific 3rd partition in f2fs with the following steps
- mount the sdcard
- copy all data with "cp -a" in the 2nde partition
- delete the second partition in ext4
- recreate a new ext4 partition with space at the end (or begining ? is there a difference using sdcard?)
- create a new partition at the end, and format it in f2fs
- create the folder and mount it
- modify the fstab as above to take it in account during boot.
and I can use it for my logs.

both solutions are working. We need to make load tests and file performance now in both solutions ;)
I can't do it right now, but I can execute some tests if you or someone have !