ev3dev/ev3dev #1107: BrickPi3: Could not resolve host

raphmc, 2018-06-04:
### What are you trying to do?
 I am trying to install the AWS IoT Pyhton SDK.

### What did you expect to happen?
It would run the script that installs it and done :)

### What actually happened?
When `Cloning into 'aws-iot-device-sdk-python'...` I get 
`fatal: unable to access 'https://github.com/aws/aws-iot-device-sdk-python.git/': Could not resolve host: github.com`.
Then I tried a simple sudo apt-get update  which it does not do either.
And I can't even ping google.com ... however, when I ping 8.8.8.8 it works.

I then added 127.0.1.1. ev3dev to `/etc/hosts` file since that was supposed to solve it says the internet. Still doesn't change the situation for me. The content of the file looks as follows:

`127.0.0.1	localhost
127.0.1.1	ev3dev	
::1	localhost ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters
172.17.0.2	0565c14fc864`

Clearly there is something wrong but I cannot figure out what it is. Help please :)

### What hardware and software are you using (including version numbers)?

I am using a  RaspberryPi3 Model B+ with a BrickPi3 and the latest Stretch snapshot that I found here: https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-05-24/

`ev3dev-sysinfo -m` gives me:

`Image file:         ev3dev-stretch-rpi2-generic-2018-05-24
Kernel version:     4.14.39-ev3dev-2.0.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=0000000059447bc4
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=00430A4B514D355934202020FF111E10
BOARD_INFO_TYPE=aux`

dlech, 2018-06-04:
How are you connecting to the Internet on the RPi? Etheret, Wi-Fi, etc.?

raphmc, 2018-06-05:
@dlech I'm connected via WIFI. I have established a SSH connection to it from my computer that works fine. I thought it might be because of the office network, so I tried another rooter and also a hotspot that I made from my mobile phone but neither worked.

raphmc, 2018-06-05:
UPDATE: I took another Raspberry Pi – Model B this time – running on Raspbian Jessie and everything works fine there (same WIFI). Don't know where the bug is on the other Raspberry Pi B+ ...

dlech, 2018-06-05:
On ev3dev you are setting up Wi-Fi using `connmanctl`, yes?

raphmc, 2018-06-05:
No, I connected a screen and a keyboard to it and did it with the visual "Brickman" interface.

dlech, 2018-06-05:
That works too. It should get the DNS server from your router. And you should be able to manually specify a DNS server though Brickman if it is not working.

raphmc, 2018-06-05:
Is there a tutorial or a link that shows how to do that? Which server to choose etc.?

dlech, 2018-06-05:
No tutorial, but is should be self-exploratory. And you could use 8.8.8.8 like you did previously.

However, if this was working on regular Raspbian with no intervention, there is something else going on. You shouldn't have to manually configure anything other than typing in your Wi-Fi passphrase.

raphmc, 2018-06-05:
yes, I think so too ... do you think it might have something to do with the BrickPi ?
Or maybe the Stretch Snapshop ?

dlech, 2018-06-05:
> do you think it might have something to do with the BrickPi ?

I doubt it.

> Or maybe the Stretch Snapshop ?

I did update the connman version recently, so maybe.

dlech, 2018-06-05:
I've tried this now with ConnMan v1.36 on RPi B+ without a problem. So, I'm guessing that you have a local network problem or you have touched a configuration file that you should not have (possibly while troubleshooting the problem).

raphmc, 2018-06-05:
okay, thanks for testing. I'll reflash the SD card tomorrow with the latest snapshot and try to connect to the internet with `connmanctl` … I'll keep you informed of any progress :)
How does your `/etc/hosts` file look?

dlech, 2018-06-05:
```
$ cat /etc/hosts
127.0.0.1	localhost
::1	localhost ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters
172.17.0.2	e28d9eb51a8d
```

raphmc, 2018-06-05:
THX ... I have read that some people add `127.0.1.1      myhostname` to it so that it works. And since I added it the `sudo` command doesn't take a minute to resolve (which it strangely does without it).
What is the `172.17.0.2   e28d9eb51a8d ` for?

dlech, 2018-06-05:
ev3dev-stretch should have `libnss-myhostname` (or something like that) so you don't need to add that line. I'm where the `172...` line comes from. Maybe from the docker container when the image was built.

raphmc, 2018-06-06:
I made it work by adding the DNS server of my rooter to `/etc/resolv.conf`.
I removed the line: `nameserver 169.254.169.254`
and replaced it with: `nameserver 192.168.0.1`
Not it works ... but not sure it will work everywhere.

raphmc, 2018-06-07:
As I thought, when I change location I have to manually add the DNS of the rooter to said `/etc/resolv.conf` file.

JorgePe, 2018-06-08:
Hi.
I'm at work so no cannot change the router. Also not using BrickPi, just EV3 but problem seems the same. Stretch image, 24 may 2018.

sudo also took a minute, had to add
127.0.1.1 ev3dev
to /etc/hosts in order to speed it up

Cannot also resolve Internet names because so edit /etc/resolv.conf to:
#nameserver 169.254.169.254
#search c.travis-ci-prod-4.internal google.internal
nameserver 8.8.8.8
nameserver 4.4.4.4
(the first two lines were already there, just commented and added the two google public DNS serversm could added my work DNS server but then would have to change again at home, I'm lazy)

I've been also using a Raspberry Pi Zero W with Raspbian stretch here at work, same Wi-Fi and it works fine so perhaps there is something in connman?

dlech, 2018-06-08:
Do older stretch images work?

JorgePe, 2018-06-08:
Have to try.
They worked in the past but my Network mates may also have changed something.
Going to try an older snapshot.

JorgePe, 2018-06-08:
Previous version I have here ("snapshot-ev3dev-stretch-ev3-generic-2018-03-16.img.xz") worked out of the box, both sudo and internet names resolution

/etc/hosts:
```
127.0.0.1	localhost
::1	localhost ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters
172.17.0.2	d75ea769beff
```

/etc/resolv.conf
```
# Generated by Connection Manager
nameserver ::1
nameserver 127.0.0.1
```

So it seems Connection Manager putting this "nameserver 169.254.169.254" is the culprit?

WasabiFan, 2018-06-11:
I'm also seeing this issue; my temporary solution was to replace `nameserver 169.254.169.254` with `nameserver 127.0.0.1` in `/etc/resolv.conf`.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-05-24
Kernel version:     4.14.38-ev3dev-2.0.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340720B
BOARD_INFO_TYPE=main
```

dlech, 2018-06-13:
The "correct" workaround is:

    sudo ln -sf /run/connman/resolv.conf /etc/

dlech, 2018-06-13:
...it turns out this was a regression made by a change in ConnMan

dlech, 2018-06-13:
Fixed in `connman 1.36-1ev3dev2`

closed 2018-06-13 by dlech

ThatCoolCoder, 2022-06-14:
For people having DNS issues in 2022, the answer to [this thread](https://github.com/ev3dev/ev3dev/issues/1504) solved my problem on an EV3.