ev3dev/ev3dev #1108: Is there any default firewall on ev3dev-stretch?

obo, 2018-06-08:
### What are you trying to do?

I am trying to run python simple http server on the robot (from http.server import BaseHTTPRequestHandler, HTTPServer; .../serve_forever).

The http server can be accessed from the robot (```telnet localhost 8080``` opens up the connection and accepts manual HTTP requests), but the port 8080 seems closed for other machines on the same wifi. Here is nmap log from my notebook, the machine from which I connect to the robot via ssh: 

```nmap -p 22,8080 192.168.1.105
Starting Nmap 7.01 ( https://nmap.org ) at 2018-06-08 21:53 CEST
Nmap scan report for robot (192.168.1.105)
Host is up (0.053s latency).
PORT     STATE  SERVICE
22/tcp open  ssh
8080/tcp closed http-proxy
```

### What did you expect to happen?

The webserver should be accessible from the whole wifi.

### What actually happened?

HTTP connection to port 8080 of the robot is refused.

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-stretch-ev3-generic-2018-04-22
Kernel version:     4.14.38-ev3dev-2.0.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535F74D2
BOARD_INFO_TYPE=main
```

I use the standard brickman wireless connection dialog. Here is the details on the network connection:

```sudo ifconfig
[sudo] password for robot: 
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 122  bytes 12056 (11.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 122  bytes 12056 (11.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=-28669<UP,BROADCAST,MULTICAST,DYNAMIC>  mtu 1500
        ether 02:16:53:5f:74:d2  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb1: flags=-28669<UP,BROADCAST,MULTICAST,DYNAMIC>  mtu 1500
        ether 22:16:53:5f:74:d2  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlxe4beeddd6996: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet 192.168.1.105  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::e6be:edff:fedd:6996  prefixlen 64  scopeid 0x20<link>
        ether e4:be:ed:dd:69:96  txqueuelen 1000  (Ethernet)
        RX packets 13148  bytes 1554598 (1.4 MiB)
        RX errors 0  dropped 47397  overruns 0  frame 0
        TX packets 6179  bytes 1061786 (1.0 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

dlech, 2018-06-08:
No, there is no firewall. Perhaps you bound your webserver to localhost instead of all interfaces.

obo, 2018-06-08:
You are correct! I haven't noticed the demo code I adapted hardcodes 127.0.0.1.

closed 2018-06-08 by obo