ev3dev/ev3dev #1109: Internet over USB not working on Ubuntu 18.04

pepijndevos, 2018-06-09:
I'm trying to update the software on the EV3 and install some things.

At first I just used the "tether" menu option with no configuration on the Ubuntu side. This got me a working SSH, but no internet.

I followed the instructions [here](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/), but Ubuntu 18.04 has a new UI that does not  have the connection sharing option, so I had to run `nm-connection-editor` to configure the network. After some rebooting and unplugging the EV3 I finally saw the "wired" option in "all network connections".

But when I try to connect it seems to give itself a bogus self-assigned IP, and I can't connect to it from my laptop. With sharing enabled, it should just transparently get an IP from my router, right? I tried painstakingly entering an IP in the correct range, to no avail.

So either I'm doing something stupid, or probably something in the Ubuntu network config changed. It's been a while, but this used to work on Ubuntu 16.04 for me.

```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-06-09:
Can you post a screenshot of the IPv4 tab in `nm-connection-editor`?

pepijndevos, 2018-06-09:
Apparently even more rebooting and unplugging fixed the problem.
I had deleted and disconnected everything, so I did it all over again to make a screenshot.
The exact steps I took now were:

1. Boot the EV3
2. *Then* connect the USB cable (this may have been the key difference)
3. Run `nm-connection-editor` to change the automatically generated profile from DHCP to shared
4. Disconnect the USB cable
5. Reconnect
6. Go to all connections and connect again

In the last step somehow the IP magically changed from bogus to 10.42.0.125, which is different from my LAN IP range, but works anyway.

If you have any idea what was going wrong before, I'd be happy to make a PR on the documentation.

closed 2018-06-09 by pepijndevos

dlech, 2018-06-09:
Tethering is really meant for sharing an Internet from the EV3 to your computer, not the other way around. It expects that your computer is not attached to any other network and it can't know if it is or not, so it is possible to get conflicting IP addresses because of this.

FYI: With ev3dev-stretch, we have IPv6 enabled, so you don't need to do anything to get a network connection. It just works using the local link address without configuring anything.

pepijndevos, 2018-06-10:
Ah, that explains some of my troubles. Initially it only seemed to show up as a CDC gadget when I turned tethering on, and did not show a "wired" option at all in "all connections", but it seems the "correct" solution is to replug it...

Looking forward to stretch.

Another random thought (maybe I should email or start a new issue): Lejos has some really extensive abstractions for all sorts of things that don't seem to exist in the ev3dev ecosystem, but does not seem as active (latest release is a two year old "beta" on Java 7). Since ev3dev has a unified API across different languages, I was wondering what the best place for such functionality would be. Python only has a Tank class, and there is a Java version that has *some* of the Lejos classes, but nothing of the same level as Lejos.

dlech, 2018-06-10:
> or start a new issue

You should do this if you don't want your thought to get lost and forgotten.

WasabiFan, 2018-06-10:
> Python only has a Tank class, and there is a Java version that has some of the Lejos classes, but nothing of the same level as Lejos.

FYI, for ev3dev-stretch in the Python library (not yet released, but available on GitHub) we have `MotorSet`, `MoveTank`, `MoveSteering` and `MoveJoystick` classes which might align with what you're looking for. I think this is likely going to need to be handled by each library/language separately for the foreseeable future, so I'd suggest that if you have suggestions for Python open an issue on that repository, and likewise for whichever Java library you use.

ifarup, 2019-04-20:
I have a similar problem with internet over USB with Ubutnu 18.04 with both jessie-ev3-generic-2017-09-14 and stretch-ev3-generic-2019-03-03. The connection seems to come up. The EV3 reports the IP address 169.254.120.209, and state goes to "State: Connected" (not "Online" as it should be according to the online docs). And the computer seems to know about it, but not being able to reach it:

$ ping ev3dev.local
PING ev3dev.local (169.254.120.209) 56(84) bytes of data.
From x1 (192.168.122.1) icmp_seq=1 Destination Host Unreachable

$ ssh robot@ev3dev.local
ssh: connect to host ev3dev.local port 22: No route to host