ev3dev/ev3dev #1110: Reboot after install kernel 2.0.2 still shows running 2.0.1

BertLindeman, 2018-06-14:
### What are you trying to do?

Boot into installed kernel 4.14.48-ev3dev-2.0.2-rpi2   on a rpi-3b-plus and rpi-zero

### What did you expect to happen?

I did NOT expect still to see Kernel version:     4.14.39-ev3dev-2.0.1-rpi2

### What actually happened?

After reboot still running Kernel version:     4.14.39-ev3dev-2.0.1-rpi2

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-04-22
Kernel version:     4.14.39-ev3dev-2.0.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=0000000020e19c66
BOARD_INFO_TYPE=main
```

### Thought:
In #987 can be a typo, I see **49** in '''This is fixed in kernel 4.14.49-ev3dev-2.0.2-ev3'''
In '/boot' I have 48 not 49 :
'''
initrd.img-4.14.48-ev3dev-2.0.2-rpi2
'''

To be sure I do have the current, did the install once more:
'''
 apt -V install  linux-image-ev3dev-rpi2
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-image-ev3dev-rpi2 is already the newest version (26).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
'''

As a sidestep installed also on an rpi-zero:
'''
The following NEW packages will be installed:
   linux-image-4.14.48-ev3dev-2.0.2-rpi (1)
. . . . . . . 
Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.14.48-ev3dev-2.0.2-rpi /boot/vmlinuz-4.14.48-ev3dev-2.0.2-rpi
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.14.48-ev3dev-2.0.2-rpi /boot/vmlinuz-4.14.48-ev3dev-2.0.2-rpi
update-initramfs: Generating /boot/initrd.img-4.14.48-ev3dev-2.0.2-rpi
Using DTB: bcm2708-rpi-0-w.dtb
Installing /usr/lib/linux-image-4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb into /boot/dtbs/4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb
Installing new bcm2708-rpi-0-w.dtb.
Installing /usr/lib/linux-image-4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb into /boot/dtbs/4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb
Taking backup of bcm2708-rpi-0-w.dtb.
Installing new bcm2708-rpi-0-w.dtb.
flash-kernel: deferring update (trigger activated)
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 4.14.48-ev3dev-2.0.2-rpi /boot/vmlinuz-4.14.48-ev3dev-2.0.2-rpi
run-parts: executing /etc/kernel/postinst.d/zz-flash-kernel 4.14.48-ev3dev-2.0.2-rpi /boot/vmlinuz-4.14.48-ev3dev-2.0.2-rpi
Using DTB: bcm2708-rpi-0-w.dtb
Installing /usr/lib/linux-image-4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb into /boot/dtbs/4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb
Taking backup of bcm2708-rpi-0-w.dtb.
Installing new bcm2708-rpi-0-w.dtb.
Installing /usr/lib/linux-image-4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb into /boot/dtbs/4.14.48-ev3dev-2.0.2-rpi/bcm2708-rpi-0-w.dtb
Taking backup of bcm2708-rpi-0-w.dtb.
Installing new bcm2708-rpi-0-w.dtb.
flash-kernel: deferring update (trigger activated)
Processing triggers for man-db (2.7.6.1-2) ...
Setting up flash-kernel (3.75ev3dev9) ...
Setting up linux-image-ev3dev-rpi (29) ...

'''

Looks OK.
After reboot this pi runs 2.0.1:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi-generic-2017-09-14
Kernel version:     4.14.39-ev3dev-2.0.1-rpi
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=9000C1
BOARD_INFO_MODEL=Raspberry Pi Zero W Rev 1.1
BOARD_INFO_SERIAL_NUM=0000000055cbb493
BOARD_INFO_TYPE=main
```

Do not know how to debug further.

dlech, 2018-06-14:
> In #987 can be a typo

this is no a typo. we are working from 3 different upstream kernels and they are rarely in sync.

> Looks OK.

Actually, it looks like `flash-kernel` never ran. Try running it manually:

    sudo flash-kernel

BertLindeman, 2018-06-14:
```
 sudo flash-kernel
Using DTB: bcm2710-rpi-3-b-plus.dtb
Installing /usr/lib/linux-image-4.14.48-ev3dev-2.0.2-rpi2/bcm2710-rpi-3-b-plus.dtb into /boot/dtbs/4.14.48-ev3dev-2.0.2-rpi2/bcm2710-rpi-3-b-plus.dtb
Taking backup of bcm2710-rpi-3-b-plus.dtb.
Installing new bcm2710-rpi-3-b-plus.dtb.
flash-kernel: installing version 4.14.48-ev3dev-2.0.2-rpi2
Will use /dev/mmcblk0p1 as boot device.
Taking backup of kernel7.img.
Installing new kernel7.img.
Taking backup of bcm2710-rpi-3-b-plus.dtb.
Installing new bcm2710-rpi-3-b-plus.dtb.

```
Looks good sofar, now reboot . . . 

Fine:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-04-22
Kernel version:     4.14.48-ev3dev-2.0.2-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=0000000020e19c66
BOARD_INFO_TYPE=main
```

dlech, 2018-06-14:
I think there is a bug in flash-kernel that if something triggers it (e.g. installing a new kernel) but the package gets updated at the same time, then it forgets about the trigger and does not run.

BertLindeman, 2018-06-14:
Good enough for me to close... There **is** a circumvention  . . .

dlech, 2018-06-14:
Agreed. We don't update flash-kernel often enough for this to cause too much of a problem and there is a workaround.

closed 2018-06-14 by dlech

dlech, 2018-07-18:
Actually, this seems to be duplicate of #612.