ev3dev/ev3dev #1112: Inconsistent time between sensor measurements

bri-davis, 2018-06-15:
I tested the time between color sensor readings for about 60 seconds using the following program:

#!/ev3/bin/env python3
from ev3dev.ev3 import *
import csv
import os.path
from time import time

color_sensor = ColorSensor(); assert color_sensor.connected, "Connect a color sensor to any sensor port"
color_sensor.mode= 'COL-REFLECT'

filepath = 'DATA/data.csv'
count = 0
while os.path.isfile(filepath):
    filepath = 'DATA/data' + str(count) + '.csv'
    count = count + 1
with open(filepath, 'a+', newline ='') as f:
        fwriter = csv.writer(f)
        fwriter.writerow(['Time Elapsed', 'dt', 'Color Value'])

button = Button()
start_time = time()
previous_time = start_time
time_elapsed = 0
dt = 0
current_time = 0
color_value = 0
while not button.any():
    current_time = time()
    color_value = color_sensor.value()
    dt = current_time - previous_time
    time_elapsed = current_time - start_time
    with open(filepath, 'a+', newline ='') as f:
        fwriter = csv.writer(f)
        fwriter.writerow([time_elapsed, dt, color_value])
    previous_time = current_time

### What did you expect to happen?
I expect that the time between sensor readings should be the same, or linearly increase as the size of the csv file increases.

### What actually happened?
The time between sensor readings is on average 0.01384898 seconds with a standard deviation of 0.00110591 seconds. You can see that the trend of the time goes up linearly as the csv file size increases, which I expected. But there is a lot of variability between the times with some being 0.012 seconds and others being 0.015 seconds. 

Additionally, what concerns me the most are the outlier points that take .02 seconds. Why are these occurring? I do not think there is something in my program causing these.

[image]

### I am using the following:
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2018-06-15:
> Why are these occurring?

Linux is not a realtime system. There a many things that could be keeping the CPU busy during this time.

Also see #324

tcwan, 2018-06-16:
To add to @dlech's comments, you're also dealing with the process scheduler in Linux in addition to any timing delays induced by the Python interpreter for garbage collection, etc.

Each process in Linux gets a given time slice on the CPU, and your program may be suspended between reading the current_time and the color_value, for example. This will cause jitter in the timing interval calculated via time_elapsed. This scheduler jitter cannot be avoided, hence the dark band of values in your graph.

The outliers can be caused by one-off events, due to normal operation of the Linux system, as well as python interpreter background tasks (e.g., garbage collection). 

As to your problem, I guess it depends on what is your goal: 
(a) to capture inputs as fast as possible at regular intervals
(b) to capture inputs at specified intervals as accurately as possible

I don't think it is possible to solve (a) well using python, or even regular Linux for the matter. You probably need to use a single process system (similar to DOS) to achieve this.
For (b) perhaps you can consider using timers to schedule input capture events.
I have not used timers in Python before, so I don't know what kind of accuracy you can achieve. However, my guess is that the shorter the interval needed the more jitter you'll encounter (depending on the system load and process scheduler behavior).

closed 2018-07-15 by dlech