ev3dev/ev3dev #1113: Shutdown down PiStorms from software

KWSmit, 2018-06-15:
Shutting down PiStorms from software (Brickman) is not working, so the only way now is by pressing the GO-button. The Mindsensors software itself does provide a script to shutdown the system, so there must be a possibillity to implement this in ev3dev (I presume).

### What hardware and software are you using (including version numbers)?
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.14.39-ev3dev-2.0.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000bc6aa21f
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=V2.08
BOARD_INFO_MODEL=PiStorms
BOARD_INFO_TYPE=aux
```

dlech, 2018-06-16:
It looks like `H` is the magic value to trigger a shutdown.

We need to call `devm_register_reboot_notifier()`. The callback should check for `action` of `SYS_POWER_OFF` and send the `H` command and return `NOTIFY_OK`, otherwise return `NOTIFY_DONE`.

This is an easy change for any aspiring kernel hackers that might be listening. :smile:

dlech, 2018-07-09:
Fix released in `ev3dev-2.1.0` kernel.

closed 2018-07-09 by dlech