ev3dev/ev3dev #1117: Disabling Brickman mouse support

obo, 2018-06-23:
I am trying to use a USB (PS/2 converted to USB, in fact) mouse as an odometer. I was surprised that Brickman supports mouse out of the box. (A pointer appears on the LCD screen and one can even control brickman with mouse.)
Can I somehow disable this? I prefer Brickman not to interfere with the mouse at all.

I also have one more question, actually a pub question rather than a real issue, which *could* be related to the unexpected mouse support. Isn't there any Easter egg controlling motors based on mouse movements? With the mouse attached to my tank robot, I tried (my modified) IR-controlled tank, but the robot totally misbehaved, it moved the other motor than I tried to control with the IR, it moved actually all three motors, there and back again, while I was trying to move just one in one direction. I had the impression that perhaps I started one movement but then some mouse ghost was trying to move the robot based on the mouse movements (unaware that the mouse is attached to the robot, causing a strange communication loop). I don't expect I will be able to reproduce this at all, but it lasted for about 10 minutes of experimenting.

### What did you expect to happen?

I was hoping to find the word 'mouse' in brickman repo, and from there to find out, how to disable it.

dlech, 2018-06-23:
Brickman uses libinput. You can create a udev rule to ignore specific devices.

https://wayland.freedesktop.org/libinput/doc/latest/udev_config.html

maybe something like this (replacing vendor/product ids with the ones for your mouse):
```
ACTION=="add", KERNEL=="event[0-9]*", ENV{ID_VENDOR_ID}=="012a", \
ENV{ID_MODEL_ID}=="034b", ENV{LIBINPUT_IGNORE_DEVICE }="1"
```

Save as `/etc/udev/rules.d/99-my-mouse.rules`

obo, 2018-06-23:
Wouldn't that prevent me from using the device in my own scripts? (I don't know udev at all.)
The way my brick sees the mouse is curiously probably this 'barcode reader':
```/dev/input/by-id/usb-13ba_Barcode_Reader-if01-event-mouse```

My initial script accessing devices is this:
```
import evdev
DEVICES = [evdev.InputDevice(fn) for fn in evdev.list_devices()]
for d in DEVICES:
    print("D: ", d)
```

The output I am getting is:
```
D:  device /dev/input/event3, name "Barcode Reader ", phys "usb-ohci-da8xx-1.4/input1"
D:  device /dev/input/event2, name "Barcode Reader ", phys "usb-ohci-da8xx-1.4/input0"
D:  device /dev/input/event1, name "EV3 Brick Buttons", phys "gpio-keys/input0"
D:  device /dev/input/event0, name "LEGO MINDSTORMS EV3 Speaker", phys ""
```

and this is the foreseen way of reading the coordinates, once I select the correct device DEVICE from DEVICES:

```
        for event in DEVICE.read_loop():
            if event.type == evdev.ecodes.EV_REL:
                if event.code == evdev.ecodes.REL_X:
                    X += event.value
                if event.code == evdev.ecodes.REL_Y:
                    Y += event.value
```

dlech, 2018-06-23:
> Wouldn't that prevent me from using the device in my own scripts?

only if you are also using libinput, which you aren't