ev3dev/ev3dev #1119: brickman dock build issues

ensonic, 2018-06-25:
the new docker based build approach is nice. Thanks! A few observations:
1. the source dir must be world readable+executable. Otherwise:
```
Successfully tagged brickman-armel:latest
d1cee0d39c9b0d444f0c792c9c47787ea5e01fae7716b66cf263a95692628e4f
CMake Error: The source directory "/src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

2. The `$BUILD_AREA` needs to be an absolute path. Otehrwise:
```
Successfully tagged brickman-armel:latest
docker: Error response from daemon: create ../brickman-jessie: "../brickman-jessie" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.
```

ensonic, 2018-06-25:
Now with those things sorted, lots of errors below.
```
Successfully tagged brickman-armel:latest
8583bddd7508ae7e3c0d782051f61ccd3219fc76e3dea97d52b917790dcdef9f
CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineSystem.cmake:175 (file):
  file Internal CMake error when trying to open file:
  /build/CMakeFiles/CMakeOutput.log for writing.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error: Could not open file for write in copy operation /build/CMakeFiles/3.0.2/CMakeSystem.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineSystem.cmake:187 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:116 (file):
  file problem creating directory: /build/CMakeFiles/3.0.2/CompilerIdC
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:39 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/share/cmake-3.0/Modules/CMakeDetermineCCompiler.cmake:108 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:116 (file):
  file problem creating directory: /build/CMakeFiles/3.0.2/CompilerIdC
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:39 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/share/cmake-3.0/Modules/CMakeDetermineCCompiler.cmake:108 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:116 (file):
  file problem creating directory: /build/CMakeFiles/3.0.2/CompilerIdC
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:39 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/share/cmake-3.0/Modules/CMakeDetermineCCompiler.cmake:108 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:529 (file):
  file Internal CMake error when trying to open file:
  /build/CMakeFiles/CMakeError.log for writing.
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/CMakeDetermineCompilerId.cmake:48 (CMAKE_DETERMINE_COMPILER_ID_VENDOR)
  /usr/share/cmake-3.0/Modules/CMakeDetermineCCompiler.cmake:108 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

-- The C compiler identification is unknown
CMake Error: Could not open file for write in copy operation /build/CMakeFiles/3.0.2/CMakeCCompiler.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /usr/share/cmake-3.0/Modules/CMakeDetermineCCompiler.cmake:172 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error: Could not find cmake module file: /build/CMakeFiles/3.0.2/CMakeCCompiler.cmake
-- Check for working C compiler: /opt/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-gcc
CMake Error at /usr/share/cmake-3.0/Modules/CMakeTestCCompiler.cmake:35 (file):
  file Internal CMake error when trying to open file:
  /build/CMakeFiles/CMakeTmp/testCCompiler.c for writing.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error at /usr/share/cmake-3.0/Modules/CMakeTestCCompiler.cmake:47 (try_compile):
  Unknown extension ".c" for file

    /build/CMakeFiles/CMakeTmp/testCCompiler.c

  try_compile() works only for enabled languages.  Currently these are:

    C

  See project() command to enable other languages.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

-- Check for working C compiler: /opt/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-gcc -- broken
CMake Error at /usr/share/cmake-3.0/Modules/CMakeTestCCompiler.cmake:58 (file):
  file Internal CMake error when trying to open file:
  /build/CMakeFiles/CMakeError.log for writing.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error at /usr/share/cmake-3.0/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler
  "/opt/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-gcc"
  is not able to compile a simple test program.

  It fails with the following output:

   

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

-- Configuring incomplete, errors occurred!
CMake Error: Cannot open file for write: /build/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /build/CMakeCache.txt
CMake Error: : System Error: Permission denied
```

ensonic, 2018-06-25:
I think the debian base is setting up users as uid=1000. This is not matching my hostsystem. I just added "-u0" to add "docker exec ..." with the downside that dist results are owned by root ...

ensonic, 2018-06-25:
And something wrong with mounting the volumes (from docker inspect)
```
        "Mounts": [
            {
                "Type": "volume",
                "Name": "brickman-jessie",
                "Source": "/var/lib/docker/volumes/brickman-jessie/_data",
                "Destination": "/build",
                "Driver": "local",
                "Mode": "z",
                "RW": true,
                "Propagation": ""
            },
            {
                "Type": "bind",
                "Source": "/home/ensonic/projects/lego/brickman",
                "Destination": "/src",
                "Mode": "",
                "RW": true,
                "Propagation": "rprivate"
            }
        ],
```

dlech, 2018-06-25:
> I just added "-u0" to add "docker exec

Why don't you use your actual user id? e.g. `-u$UID`

dlech, 2018-06-25:
> And something wrong with mounting the volumes

what is wrong?

ensonic, 2018-06-26:
> because I can't write with that UID (since my UID ()uid=141164(ensonic) is not the same that debian assumes). The debian base would need to import my UID, but it just hard-codes one.

ensonic, 2018-06-26:
> what is wrong?
build is mounted to /var/lib/docker/volumes/brickman-jessie/_data and not to the local "brickman-jessie/" dir that I passed on the command-line for $BUILD_AREA

ensonic, 2018-06-26:
One more thing. When I build as the documentation suggest, my binary is large (the brickman.bak is the orifinal one):
```
$ ls -al /usr/sbin/brickman*
-rwxr-xr-x 1 robot robot 3399947 Jun 26 10:06 /usr/sbin/brickman
-rwxr-xr-x 1 root  root   851288 Jan 14  2017 /usr/sbin/brickman.bak
$ strip /usr/sbin/brickman
$ ls -al /usr/sbin/brickman*
-rwxr-xr-x 1 robot robot 1310544 Jun 26 11:33 /usr/sbin/brickman
-rwxr-xr-x 1 root  root   851288 Jan 14  2017 /usr/sbin/brickman.bak
```
Would be nice to add to the instruction how to do a release build.

dlech, 2018-06-26:
> Would be nice to add to the instruction how to do a release build.

set `CMAKE_BUILD_TYPE:Release`

dlech, 2018-06-26:
I just created user `ensonic` with UID:GID `141164:141164` on my host system. Then ran as that user with the change above and was able to successfully build brickman.

ensonic, 2018-06-26:
> set CMAKE_BUILD_TYPE:Release
wouldn't one need to change the dockerfile for that? This is passed to cmake, right?

dlech, 2018-06-26:
Nope, it can be changed after the fact. I usually just edit the `CMakeCache.txt` file in `build-armel/` directly. I'm sure there is a way to do it with the cmake command as well, but that would require reading the docs.

dlech, 2018-06-26:
If we installed `cmake-curses-gui` in the docker image, then this would work as well:

    docker exec -it brickman_armel ccmake /src /build